Clon1998/mobileraker #123: No IOS notifications

redlinebikeparts, 2023-03-06:
I'm not getting any notifications through the IOS app. 

I understand I won't get them when I'm outside my local network but I'm not getting them when I'm inside my local network. 

I've tried uninstalling and reinstalling the app. Manually deactivating/ reactivating notifications in both ios settings and app settings

Clon1998, 2023-03-06:
Hey,
I assume you don't have the companion app installed that is required to get notifications to work, do you?

https://github.com/Clon1998/mobileraker_companion

As soon as you installed it, restart the PI and the app and you should receive notifications according to your push notification settings defined in the app even if you only connected to cellular.

redlinebikeparts, 2023-03-08:
Hi I had installed the companion app. I just checked the config file and it doesn't look the same as your examples so I have made some changes and will see what happens

Clon1998, 2023-03-14:
> Hi I had installed the companion app. I just checked the config file and it doesn't look the same as your examples so I have made some changes and will see what happens 

I assume that notifications are working for you now?
Therefore, I will close the issue. If it's not working please comment/reopen it.