Clon1998/mobileraker #125: I have been using mobile Raker locally and absolutely love it. Recently I started connecting to it Via Tail Scale VPN with great success! Is it possible to have two Ip's (local and VPN) assigned to one printer.

Tmallard, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Naemion, 2023-03-20:
So if you have a pc or something that is on all the time you can set that up in TailScale as a subnet router then through that you can use the same ip for the printer whether you on the local IP or the VPN. It is allowing your phone to access the entire home network instead of just the device you're connected to.

Clon1998, 2023-03-20:
> So if you have a pc or something that is on all the time you can set that up in TailScale as a subnet router then through that you can use the same ip for the printer whether you on the local IP or [...]

That just sounds like any other VPN service/solution. I am not to sure why your setup does require two different ips. By default it's sufficient to provide the local network ip of your machine to the app. Connect your phone via any VPN tunnel to your home network. Afterwards the entire network traffic is tunneld to your home network including any ARP requests.

closed 2023-04-21 by Clon1998