Clon1998/mobileraker #132: Not receiving any notifications on iOS

chronocentric, 2023-03-26:
When I first switched to Klipper, I installed it on a linux virtual machine using KIAUH, then added Mobileraker companion and notifications were working fine. A short time later I managed to get my hands on a Raspberry Pi and did a fresh install using Mainsail OS, and again added Mobileraker, but notifications have never worked on the new Pi. I have tried uninstalling and re-installing the app on my iPhone and it did not help. App permissions are correctly set. I have updated Mobileraker through Mainsail, and (based on a warning notification in Mainsail) added it to Moonraker so that it can control the Mobileraker service. Still no notifications. Any ideas? Thanks!

Clon1998, 2023-03-26:
> When I first switched to Klipper, I installed it on a linux virtual machine using KIAUH, then added Mobileraker companion and notifications were working fine. A short time later I managed to get my hands on a Raspberry Pi and did a fresh install using Mainsail OS, and again added Mobileraker companion, but notifications have never worked on the new Pi. I have tried uninstalling and re-installing the app on my iPhone and it did not help. App permissions are correctly set. I have updated Mobileraker companion through Mainsail, and (based on a warning notification in Mainsail) added it to Moonraker so that it can control the Mobileraker service. Still no notifications. Any ideas? Thanks!

Hey can you provide me the logs of the companion (Currently still located under `/tmp/mobileraker.log`.
In addition please post the output of the following HTTP request `http://<YOUR-PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm` (make sure to replace <YOUR-PRINTER-IP> with the correct IP).

chronocentric, 2023-03-26:
Here is the HTTP output:

<html><body>

namespace | "mobileraker"
-- | --
key | "fcm"
value |  
b23df499-9113-404d-977b-59dd81fe8fb2 |  
created | "2023-02-23T15:05:26.920733"
lastModified | "2023-02-23T15:05:26.920735"
fcmToken | "dfnDJJzsOk1pmIAlzLr0vr:APA91bF1ID-tNyvjw7neruHCfg0DF6HoEfYElPNpW_i1O61AgpgCEA9ZgATKoI1BlQVD8tsNODHxV0exrrFNy_u5R5fZD97inx_seAnlQ9ZsjkTKraivnP_tYe3FUuEoWrDPE7kWsy7S"
machineName | "Dragon Forge"
language | "en"
settings |  
created | "2023-02-23T15:05:26.920733"
lastModified | "2023-02-24T14:55:17.204042"
progress | 0.25
states |  
0 | "printing"
1 | "paused"
2 | "complete"
3 | "error"
4 | "standby"
snap | null
db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99 |  
created | "2023-02-25T00:24:06.996649"
lastModified | "2023-02-25T00:24:06.996668"
fcmToken | "fGwFFS1pZUYDoSdyhn7Cc0:APA91bFheHnQ5RaHh8998whLeFARoCzgxY-ulT1ofL3KuT6PHpkQ23qw7vmmXLR_MM9kbG4WK-vBQBXI6HMpp0ramjRpKlQXZS5VgKQCnfjoO_5lJ0PXE9DYRwnBbAXeufL-YuUWLluD"
machineName | "Dragon Forge"
language | "en"
settings |  
created | "2023-02-25T00:24:06.996651"
lastModified | "2023-03-25T12:57:56.685610"
progress | 0.1
states |  
0 | "paused"
1 | "error"
2 | "complete"
3 | "standby"
4 | "printing"
snap | null

</body>
</html>

I can't find the logs directory on my pi, probably because Mainsail OS sets up a username pi and not root? I'm a linux newb...

Clon1998, 2023-03-26:
I will need to use a ssh terminal in order to download the log. However, the result of the request is already good info.
Are you using more than one device with the printer and mobileraker? Or just a single device?
If the later is the case, you will need to reset the database entry.
I will post a instructions later on how to do that.

chronocentric, 2023-03-26:
I am using one device with one printer. I was wondering if it was a database issue considering I moved from one instance of klipper to another.

Clon1998, 2023-03-26:
> I am using one device with one printer. I was wondering if it was a database issue considering I moved from one instance of klipper to another.

We will see :).
I hope you are on windows, otherwise let me know and I will look up the command.
1. Open your cmd (e.g. via WINDOWS-KEY + R, and enter `cmd`)
2. Run the following command `powershell -command 'Invoke-WebRequest -UseBasicParsing -Method DELETE -Uri http://192.168.178.135/server/database/item?namespace=mobileraker"&"key=fcm'`
3. Restart the Pi and also force close mobileraker on your phone and open it again as soon as the printer is back up
4. You can test if the notifications are working by issuing `M117 $MR$:Hey I am a notification` in the printer's console via either Fluidd/Mainsail

chronocentric, 2023-03-27:
I tried running the command, substituting my pi's IP address, and can't get it to work. It just prints the command back but doesn't seem to actually do anything.

chronocentric, 2023-03-29:
Is there any other way to reset the database?

Clon1998, 2023-03-29:
> Is there any other way to reset the database?

Yes there is. However I don't think the database is the problem here. Can you make sure to update the companion to the latest version and also run the fix log script (`~/mobileraker_companion/scripts/fix-log-location.sh`) in order for you to see the logs directly in mainsail. Afterwards please share them with me.

chronocentric, 2023-03-30:
Updated...here are the logs...notifications worked briefly after I ran the fix log script...

-------------------MobilerakerCompanion Log Start--------------------
Git Version: v0.3.0-1-gf17e126
2023-03-29 22:18:36,552 root INFO - MobilerakerCompanion version: v0.3.0-1-gf17e126
2023-03-29 22:18:36,553 root INFO - Passed config file is: /home/pi/Mobileraker.conf
2023-03-29 22:18:36,554 root INFO - Checking mobileraker_companion dir
2023-03-29 22:18:36,554 root INFO - Checking klipper_config dir
2023-03-29 22:18:36,554 root INFO - Checking printer_data/config dir
2023-03-29 22:18:36,555 root INFO - Checking user dir
2023-03-29 22:18:36,555 root WARNING - No config file was found, using default fallback values!
2023-03-29 22:18:36,555 root INFO - Read 1 printer config sections
2023-03-29 22:18:36,605 root INFO - Main section read, language:"en", timezone:"America/Chicago", eta_format:"%d.%m.%Y, %H:%M:%S"
2023-03-29 22:18:36,607 mobileraker.jrpc INFO - Trying to connect to: ws://127.0.0.1:7125/websocket api key <NO API KEY>
2023-03-29 22:18:36,615 mobileraker.jrpc INFO - WebSocket connected
2023-03-29 22:18:36,616 mobileraker._Default INFO - Fetching printer Objects Try#0
2023-03-29 22:18:36,619 mobileraker._Default INFO - Received Server Info
2023-03-29 22:18:36,620 mobileraker._Default INFO - Querying printer Objects
2023-03-29 22:18:36,702 mobileraker._Default INFO - Subscribing to printer Objects
2023-03-29 22:18:36,707 mobileraker._Default INFO - Got Cfgs!
2023-03-29 22:18:36,708 mobileraker._Default INFO - Got configs: [<dtos.mobileraker.notification_config_dto.DeviceNotificationEntry object at 0x7fb93ebd60>, <dtos.mobileraker.notification_config_dto.DeviceNotificationEntry object at 0x7fb93ebf70>]
2023-03-29 22:18:36,708 mobileraker._Default INFO - Evaluate for machineID b23df499-9113-404d-977b-59dd81fe8fb2, snap: standby 0, cfg: 25  ['printing', 'paused', 'complete', 'error', 'standby']
2023-03-29 22:18:36,708 mobileraker._Default INFO - Notifications for machineID: b23df499-9113-404d-977b-59dd81fe8fb2:0, state:False, proggress:False
2023-03-29 22:18:36,708 mobileraker._Default INFO - Evaluate for machineID db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99, snap: standby 0, cfg: 10  ['paused', 'error', 'complete', 'standby', 'printing']
2023-03-29 22:18:36,709 mobileraker._Default INFO - Notifications for machineID: db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99:0, state:False, proggress:False
2023-03-29 22:18:36,709 mobileraker._Default INFO - ---- Done wiht evaluate notification Task! ----
2023-03-29 22:18:36,959 mobileraker._Default INFO - Got Cfgs!
2023-03-29 22:18:36,959 mobileraker._Default INFO - Got configs: [<dtos.mobileraker.notification_config_dto.DeviceNotificationEntry object at 0x7fb93ebd60>, <dtos.mobileraker.notification_config_dto.DeviceNotificationEntry object at 0x7fb9412040>]
2023-03-29 22:18:36,959 mobileraker._Default INFO - Evaluate for machineID b23df499-9113-404d-977b-59dd81fe8fb2, snap: standby 0, cfg: 25  ['printing', 'paused', 'complete', 'error', 'standby']
2023-03-29 22:18:36,960 mobileraker._Default INFO - Notifications for machineID: b23df499-9113-404d-977b-59dd81fe8fb2:0, state:False, proggress:False
2023-03-29 22:18:36,960 mobileraker._Default INFO - Evaluate for machineID db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99, snap: standby 0, cfg: 10  ['paused', 'error', 'complete', 'standby', 'printing']
2023-03-29 22:18:36,960 mobileraker._Default INFO - Notifications for machineID: db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99:0, state:False, proggress:False
2023-03-29 22:18:36,960 mobileraker._Default INFO - ---- Done wiht evaluate notification Task! ----
2023-03-29 22:19:45,409 mobileraker._Default INFO - Evaluating m117 notifications!
2023-03-29 22:19:45,421 mobileraker._Default INFO - Got Cfgs!
2023-03-29 22:19:45,422 mobileraker._Default INFO - Got M117/Custom: Hey I am a notification: User Notification - Hey I am a notification
2023-03-29 22:19:45,423 mobileraker._Default INFO - Updating snap.m117 in FCM Cfg for b23df499-9113-404d-977b-59dd81fe8fb2 c6b9af31793c173822c516149eaa7835704d1112051db1a72622479b0d391c68
2023-03-29 22:19:45,425 mobileraker._Default INFO - Got Cfgs!
2023-03-29 22:19:45,426 mobileraker._Default INFO - Got configs: [<dtos.mobileraker.notification_config_dto.DeviceNotificationEntry object at 0x7fb93ebf10>, <dtos.mobileraker.notification_config_dto.DeviceNotificationEntry object at 0x7fb94123a0>]
2023-03-29 22:19:45,427 mobileraker._Default INFO - Evaluate for machineID b23df499-9113-404d-977b-59dd81fe8fb2, snap: standby 0, cfg: 25  ['printing', 'paused', 'complete', 'error', 'standby']
2023-03-29 22:19:45,428 mobileraker._Default INFO - Notifications for machineID: b23df499-9113-404d-977b-59dd81fe8fb2:0, state:False, proggress:False
2023-03-29 22:19:45,428 mobileraker._Default INFO - Evaluate for machineID db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99, snap: standby 0, cfg: 10  ['paused', 'error', 'complete', 'standby', 'printing']
2023-03-29 22:19:45,429 mobileraker._Default INFO - Notifications for machineID: db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99:0, state:False, proggress:False
2023-03-29 22:19:45,429 mobileraker._Default INFO - ---- Done wiht evaluate notification Task! ----
2023-03-29 22:19:45,436 mobileraker._Default INFO - Got M117/Custom: Hey I am a notification: User Notification - Hey I am a notification
[... cut ...]
2023-03-29 22:53:20,464 mobileraker._Default INFO - Notifications for machineID: b23df499-9113-404d-977b-59dd81fe8fb2:0, state:False, proggress:False
2023-03-29 22:53:20,464 mobileraker._Default INFO - Evaluate for machineID db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99, snap: standby 0, cfg: 10  ['paused', 'error', 'complete', 'standby', 'printing']
2023-03-29 22:53:20,464 mobileraker._Default INFO - Notifications for machineID: db6b8e4a-0e35-4e4e-a0eb-997e6b5c1a99:0, state:False, proggress:False
2023-03-29 22:53:20,464 mobileraker._Default INFO - ---- Done wiht evaluate notification Task! ----

chronocentric, 2023-03-30:
Update: It looks like notifications are working! Not sure what fixed it, I was thinking maybe the companion was not getting started correctly? Anyway, thanks for you help and thanks for the great app!

closed 2023-03-30 by chronocentric

Clon1998, 2023-03-30:
> Update: It looks like notifications are working! Not sure what fixed it, I was thinking maybe the companion was not getting started correctly? Anyway, thanks for you help and thanks for the great app!

I am glad that it is working now!
I also think the companion was not running correctly or ran into an endless loop that would have required a restart, since in v0.2.x I fixed a bug related to an endless loop.