Clon1998/mobileraker #133: Duplicate Notifications Issue

sabaatworld, 2023-03-26:
**Describe the bug**
Recently, I've installed the companion app to enable notification. Now notifications seems to work but I always get two notifications for the same event. One says the name of my printer and the other says "Printer LAN". Is this a known issue? Are there any workarounds?

**To Reproduce**
Steps to reproduce the behavior:
1. Install and connect Mobileraker iOS app to Klipper.
2. Install companion app on Klipper host.
3. Start print to receive notification.

**Expected behavior**
I would have expected a single notification for my printer "MK3S+" and nothing from "Printer LAN".

**Screenshots**
Screenshots attached for notifications and application configuration.

**Desktop (please complete the following information):**
 - OS: iOS
 - Version 2.2.1-143

**Smartphone (please complete the following information):**
 - Device: iPhone 12 Pro
 - OS: iOS 16.6

**Mobileraker-Version
v0.2.1-53-gcdb01c5

**Additional context**
N/A

[image]
[image]

Clon1998, 2023-03-26:
> 

Hey, it seems like there is an old token still registered for your phone. 
Can you provide me with the output of this request: `http://<YOUR-PRINTER-IP>/server/database/item?namespace=mobileraker&key=fcm`
Also, are you using mobileraker with only a single Tablet/Phone?

sabaatworld, 2023-03-26:
Makes sense. I'm using it with one printer but had previously installed it for another printer. Do you know how to prevent this issue from happening in the future?

```
{
  "result": {
    "namespace": "mobileraker",
    "key": "fcm",
    "value": {
      "9f414c1e-274a-4b7c-bbaf-dccc89158b71": {
        "created": "2023-03-03T01:24:23.482617",
        "lastModified": "2023-03-03T01:24:23.482632",
        "fcmToken": "fLNtBqfIM0ITpjsYrV6A7a:APA91bE9GeS-79KcO81vDevGXKsSYvbcLLD30LQ-AKwbB_W6vEWi_WMG_y89gmh3bzhEdELEKJ3dzmGdzPg7ff9-OPHd6LS7e767xqw3bAAkA8lYnB1E9-Bn2V7DwVJLmzpAsEDr02r-",
        "machineName": "Printer LAN",
        "language": "en",
        "settings": {
          "created": "2023-03-03T01:24:23.482619",
          "lastModified": "2023-03-03T01:24:23.482618",
          "progress": 0.25,
          "states": [
            "standby",
            "printing",
            "paused",
            "complete",
            "error"
          ]
        },
        "snap": {
          "progress": 0,
          "state": "complete",
          "m117": ""
        }
      },
      "5cd7134d-17ef-4062-a2ab-307db2db3844": {
        "created": "2023-03-03T01:26:49.964140",
        "lastModified": "2023-03-03T01:26:49.964154",
        "fcmToken": "fLNtBqfIM0ITpjsYrV6A7a:APA91bE9GeS-79KcO81vDevGXKsSYvbcLLD30LQ-AKwbB_W6vEWi_WMG_y89gmh3bzhEdELEKJ3dzmGdzPg7ff9-OPHd6LS7e767xqw3bAAkA8lYnB1E9-Bn2V7DwVJLmzpAsEDr02r-",
        "machineName": "MK3S+",
        "language": "en",
        "settings": {
          "created": "2023-03-03T01:26:49.964142",
          "lastModified": "2023-03-26T07:05:57.853172",
          "progress": -1,
          "states": [
            "printing",
            "error",
            "complete",
            "paused",
            "standby"
          ]
        },
        "snap": {
          "progress": 0,
          "state": "complete",
          "m117": ""
        }
      }
    }
  }
}
```

Clon1998, 2023-03-26:
Exactly what I thought, the same phone with two different IDs.

I assume you are capable of doing an HTTP DELETE do clear all data: `DELETE http://<YourPrinter>/server/database/item?namespace=mobileraker&key=fcm`, afterward all you need to do is foreclose the app and open it again.

> Makes sense. I'm using it with one printer but had previously installed it for another printer. Do you know how to prevent this issue from happening in the future?

Currently, the fastest and easiest idea is to add "Reset Notification Devices" to the app, since the API of the firebase lib does not offer me any function to get all of the old IDs of a device. 
Another option might be to restrict the lifetime of these config entries to e.g. 7 days. After that time the plugin could disable them/ignore them during a notification creation process. Once the app on a device is opened again, the config could be reactivated. Alternatively, I could also clear the entry after 7 days without a reset/keepalive. However, this will also delete the notification settings...

Anyways, I will need to look into that a bit more, since the IDs on my devices never actually change <.<

Clon1998, 2023-03-26:
Double checked the JSON you provided me.
It seems like it contained an old Printer that the app isn't actually using anymore. Therefore, I added some code that removes notificationSetting entries of the device if the machine isn't present on the device anymore in [565ee99](https://github.com/Clon1998/mobileraker/commit/565ee993d5e6e2824d686af866105831dee71069)

sabaatworld, 2023-03-26:
HTTP DELETE did fix the issue.

Thanks for pushing out the fix!

commit 565ee993 (2023-03-26): Fixed #133 - Added check to prevent old notificationSettings to be present
commit 7265196a (2023-04-11): Fixed #133 - Added check to prevent old notificationSettings to be present
commit 35812851 (2023-04-20): Fixed #133 - Added check to prevent old notificationSettings to be present
commit 7a101340 (2025-04-13): fix: Microsteps in no longer a required field.

By default klipper REQUIRES that the microsteps field is present for any stepper config. However, some custom forks do not obey to this rule. Since MR currently does not use this information I made it optional.

fixes #133
closed 2023-03-26 by Clon1998