Clon1998/mobileraker #135: Flutter error

olivier37520, 2023-03-28:
I found an error, screenshots provided.

Step to reproduce:
1 Open slide panel
2 click on the current printer settings
3 click on the back button on top left 
[image]

Clon1998, 2023-03-28:
Can you provide me with the app version that is shown at the bottom of the app settings page?

olivier37520, 2023-03-28:
[image]

olivier37520, 2023-03-28:
[image]

Clon1998, 2023-03-28:
Hey thanks for the version.
Please make sure you update to 2.2.1 and let me know if the issue is resolved or not

olivier37520, 2023-03-29:
Same issue with build from source.

sakib-shaikh-github, 2023-03-29:
I could only understand the PrimaryScrollController, thing you need wrap that around _builder_ so it could get an context to build.

Clon1998, 2023-03-29:
> I could only understand the PrimaryScrollController, thing you need wrap that around _builder_ so it could get an context to build.

Nope. The bug is caused by the dependency that provides the reordablewrap.

See: https://github.com/hanshengchiu/reorderables/pull/174