Clon1998/mobileraker #138: Layer progress and number of layers incorrect

sammcj, 2023-04-03:
**Describe the bug**

I also noticed that at times the current and number of layers appears to be incorrect.

Possibly related - the estimates in Mobileraker are often quite different to what's shown in Mainsail and the Slicer (related - #43).

t - I suspect these two things are related.

**To Reproduce**
Steps to reproduce the behaviour:

1. When printing check the current layer in Mainsail. (example gcode provided)
2. Check the current layer layer in Mobileraker.
3. Note that these widely differ.

This print had a filament changed mid way through - I'm wondering if that throws Mobile raker out perhaps?

**Expected behaviour**

Mobileraker should show the same current and total layer numbers as Mainsail.

**Screenshots**

Mainsail:

- Current layer: 313
- Total layers: 718
- Elapsed: 1h 10m
- ETA: 20:45
- Estimate: 1:16:20
- Slicer: 2:11:35
- Total: 1:10:34

[image]

Mobileraker:

- Current layer: 168
- Total layers: 496
- Elapsed: 1h 10m
- ETA: 20:18
- Estimate: Not shown
- Slicer: Not shown (but you can see it in the GCODE file name being 3h 22m)
- Total: Not shown

[image]

Job history after print completed:

- Estimated time: 3h 21m
- Print time: 2h 30min

(Clearly Klipper/Mainsail and my slicer were all very wrong!)

[image]

**Smartphone (please complete the following information):**
 - Device: iPhone 14 Pro
 - OS: 16 (latest)
 - Browser: Safari
 - Version: Latest

**Mobileraker-Version

2.2.1-143

**Additional context**

- Klipper: v0.11.0-159-gfec7ddd8
- Slicer: PrusaSlicer 2.6.0-Alpha6
	- Gcode Flavour: Klipper
- Printer: Ender 5 S1
- [Example GCODE](https://github.com/Clon1998/mobileraker/files/11133630/FOR.REVISION.2.of.ENDER.5.S1.EL.PABLO.for.Ender.5.S1.by.NBR_3h22m_0.24mm_206C_PLA_ENDER5S1.gcode.zip)

<details> Mobileraker debug logs 

<summary> mobile logs </summary>

```
[[38;5;12m💡 Started initializeAvailableMachines[0m]
[[38;5;12m💡 Selecting machine rockpi[0m]
[[38;5;196m⛔ machineProvider creation STARTED 0e018f10-015d-4f56-b041-52408907c640[0m]
[[38;5;196m⛔ machineProvider creation STARTED 2535017e-398c-4af3-96e2-08879f20dc0f[0m]
[[38;5;196m⛔ machineProvider creation DONE 0e018f10-015d-4f56-b041-52408907c640[0m]
[[38;5;196m⛔ machineProvider creation DONE 2535017e-398c-4af3-96e2-08879f20dc0f[0m]
[[38;5;12m💡 initialized all machineProviders[0m]
[[38;5;12m💡 Init for rockpi(2535017e-398c-4af3-96e2-08879f20dc0f)[0m]
[[38;5;12m💡 [360883873]Trying to connect to ws://1<redacted>/websocket[0m]
[[38;5;12m💡 [360883873-ws://<redacted>/websocket] ClientState.disconnected ➝ ClientState.connecting[0m]
[[38;5;12m💡 Finished initializeAvailableMachines[0m]
[[38;5;196m⛔ ääää AsyncData<ClientState>(value: ClientState.disconnected)[0m]
[[38;5;196m⛔ ääää AsyncData<ClientState>(value: ClientState.disconnected)[0m]
[[38;5;12m💡 Notifications are permitted: true[0m]
[[38;5;12m💡 Token from FCM <redacted>:<redacted>[0m]
[[38;5;12m💡 Device's FCM token: <redacted>:<redacted>[0m]
[[38;5;196m⛔ ääää AsyncData<ClientState>(value: ClientState.disconnected)[0m]
[[38;5;12m💡 Token from FCM <redacted>:<redacted>[0m]
[[38;5;12m💡 Device's FCM token: <redacted>:<redacted>[0m]
[[38;5;12m💡 [301866142]Trying to connect to ws://<redacted>/websocket[0m]
[[38;5;12m💡 [301866142-ws://<redacted>/websocket] ClientState.disconnected ➝ ClientState.connecting[0m]
[[38;5;196m⛔ ääää AsyncData<ClientState>(value: ClientState.disconnected)[0m]
[[38;5;12m💡 Token from FCM <redacted>:<redacted>[0m]
[[38;5;12m💡 [301866142-ws://<redacted>/websocket] ClientState.connecting ➝ ClientState.connected[0m]
[[38;5;12m💡 >>>Fetching Server.Info[0m]
[[38;5;12m💡 >>>Fetching Printer.Info[0m]
[[38;5;12m💡 <<<Received Server.Info[0m]
[[38;5;12m💡 Refreshing printer for uuid: 2535017e-398c-4af3-96e2-08879f20dc0f[0m]
[[38;5;12m💡 >>>Querying printers object list[0m]
[[38;5;12m💡 Jrpc Client of rockpi(ws://<redacted>/websocket) is connected, can Setup FCM on printer now![0m]
[[38;5;12m💡 Getting fcm.2535017e-398c-4af3-96e2-08879f20dc0f[0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 <<<Received Printer.Info[0m]
[[38;5;12m💡 Current FCMConfig in MoonrakerDB: DeviceFcmSettings{machineId: e4RvtSCiskzPmQkTEJJz6H:APA91bE8_o_QKACJ7b7Ah4KgtqBJBU7uNx1SWaAlXDK1vlxd60TfD1n8w3TIaQ-xijqX0747RZJ8reaay8de0PPcLMgUtxG2Mzg6z6QRcEjft0jBz-72DhwopSWXQ8qAym-6Bf7JMFEG, machineName: rockpi, language: en, settings: NotificationSettings{progress: 0.5, states: {paused, complete, error, standby}}}[0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 <<<Received printer objects list![0m]
[[38;5;12m💡 >>>Querying Printer Objects![0m]
[[38;5;12m💡 <<<Received queried printer objects[0m]
[[38;5;12m💡 Fetching cached temperature store data[0m]
[[38;5;12m💡 Received cached temperature store for [temperature_sensor Rock_pi, heater_bed, extruder][0m]
[[38;5;12m💡 Subscribing printer objects for ws-updates![0m]
[[38;5;12m💡 Updating Default Macros for "rockpi(ws://<redacted>/websocket)"![0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 Start MJPEG - targFps: 15 - [http://<redacted>/webcam2/?action=stream[0m](http://<redacted>/webcam2/?action=stream%1B[0m)]
[[38;5;12m💡 Getting meta for file: `FOR REVISION 2 of ENDER 5 S1 EL PABLO for Ender 5 S1 by NBR_3h22m_0.24mm_206C_PLA_ENDER5S1.gcode`[0m]
[[38;5;12m💡 Adding 5 new macros to default group![0m]
[[38;5;12m💡 Refreshing ClientType.local was PULL to REFRESH[0m]
[[38;5;12m💡 Refreshing printer for uuid: 2535017e-398c-4af3-96e2-08879f20dc0f[0m]
[[38;5;12m💡 >>>Querying printers object list[0m]
[[38;5;208m⚠️ -DISPOSED printerSelected[0m]
[[38;5;12m💡 <<<Received printer objects list![0m]
[[38;5;12m💡 >>>Querying Printer Objects![0m]
[[38;5;208m⚠️ Received statusUpdate before a printer was parsed initially![0m]
[[38;5;12m💡 <<<Received queried printer objects[0m]
[[38;5;12m💡 Fetching cached temperature store data[0m]
[[38;5;12m💡 Received cached temperature store for [temperature_sensor Rock_pi, heater_bed, extruder][0m]
[[38;5;12m💡 Subscribing printer objects for ws-updates![0m]
[[38;5;12m💡 Updating Default Macros for "rockpi(ws://<redacted>/websocket)"![0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 App in AppLifecycleState.inactive[0m]
[[38;5;12m💡 App forgrounded[0m]
[[38;5;12m💡 Refreshing selectedPrinter...[0m]
[[38;5;12m💡 Start MJPEG - targFps: 15 - [http://<redacted>/webcam2/?action=stream[0m](http://<redacted>/webcam2/?action=stream%1B[0m)]
[[38;5;12m💡 App in AppLifecycleState.inactive[0m]
[[38;5;12m💡 App backgrounded[0m]
[[38;5;12m💡 Stop MJPEG[0m]
[[38;5;12m💡 App in AppLifecycleState.inactive[0m]
[[38;5;12m💡 App forgrounded[0m]
[[38;5;12m💡 Refreshing selectedPrinter...[0m]
[[38;5;12m💡 Start MJPEG - targFps: 15 - [http://<redacted>/webcam2/?action=stream[0m](http://<redacted>/webcam2/?action=stream%1B[0m)]
[[38;5;12m💡 Stop MJPEG[0m]
[[38;5;12m💡 _AdaptiveStreamManager DISPOSED[0m]
[[38;5;208m⚠️ -DISPOSED printerSelected[0m]
</details>

---

p.s. - love the app, thank you so much for creating it!

Clon1998, 2023-04-10:
Hey,
for me it is working more than fine (See attached screenshots).
However, I think the difference is caused by the fact that Mobileraker does not use the same information as fluidd/mainsail does. I will have another look into that but afaik I just implemented the ETA calculation the same way as Fluidd did.
Another idea might be that my print was to short to show the large differences.
[image]
[image]
[image]

sammcj, 2023-04-20:
Sorry I didn't see this - that's interesting - I suspect it must be something relating to my slicing so I'll close this for now, thank you for your efforts!

fuzelet, 2023-07-15:
Not to bring up a closed topic, but wanted to bring this up again as I myself have been noticing this using Mainsail and Mobileraker. I can open a new thread or let it go based on information above. I also understand the ETA differences, I'm interested in the layer count being so far off. The website (mainsail) and the mobileraker app show different layer counts, etc. Mainsail is 100% correct in terms of layer count as it aligns with the actual print. Not sure why mobileraker is off by 18 layers.

Mainsail
[image]

Mobileraker 
[image]

fuzelet, 2023-07-16:
I would like to add, I was watching the end of a print and started noticing the app would increase then decrease the layer over and over again as it slowly moved up, etc. As you can see from the layer status from gcode, I'm already past the end layer count that the application said was the end.

https://github.com/Clon1998/mobileraker/assets/70331495/723f8f6e-67c3-439c-88b6-6a6093edf386

Clon1998, 2023-07-17:
Hey,
it has been a while since I implemented the layer display. I compared my implementation with the others and it seemes like Klipper/moonraker offers new sources to get the current and max layer.
In contrast Mobileraker is still only using the Z height of the print alongside the layer height reported by Klipper to calculate the max and current layer. I will update the code to ensure the layer height is more consistent with the other UIs. Once I am done with that, do you mind testing that for me? I don't have access to my printer for the next two months.

fuzelet, 2023-07-17:
> Hey, it has been a while since I implemented the layer display. I compared my implementation with the others and it seemes like Klipper/moonraker offers new sources to get the current and max layer. [...]

I have no issues helping out where I can.

Clon1998, 2023-07-17:
@fuzelet please give it a try: 
https://testflight.apple.com/join/h9JwuyR0

(Let me know if you are on Android and need an APK)

fuzelet, 2023-07-17:
> @fuzelet please give it a try: https://testflight.apple.com/join/h9JwuyR0
> 
> (Let me know if you are on Android and need an APK)

I am on Android.

joschiv1977, 2023-07-18:
Same problem here. Tested also with your Beta on iOS. 

[image]
[image]

Clon1998, 2023-07-18:
@fuzelet I will build an apk later today and upload it to GitHub.

@joschiv1977, I probably should have mentioned the version number too. Please make sure you're version is 2.5.0-222. in addition I only changed the current and max layer display. The progress might still differ from mainsail.

joschiv1977, 2023-07-18:
[image]

joschiv1977, 2023-07-18:
Here you got all 4 settings in Moonraker and Mobileraker

[image]
[image]

Clon1998, 2023-07-18:
> Here you got all 4 settings in Moonraker and Mobileraker. My setting is slicer M73
> 
> 
> 
> [image]
> 
> [image]
> 
> 

Thanks for the detailed answer.
As I said I am currently not trying to match mainsails print progress and eta calculation. However, I try to give you some insides regarding the progress and eta calculation:
There are different ways on how the progress and eta can be calculated. Mobileraker is currently only using a quiet simple approach while Mainsail allows the users to select the actual method. In addition there is not a single true way of calculating these values since they are estimates.

My change in the TestFlight version only was aimed at matching the layer/max layer display between mobileraker and other UIs.
Judging by your screenshots I seem to have reached that goal (Being within 1 layer is good enough for me)

fuzelet, 2023-07-23:
Do we know if the changes fixes an issue with notifications? I never got an android apk to test the earlier change.

[image]

I just started printing this file and the printer is still warming up but the application already says it's 3 layers done and 11% completed. This looks to mess up the notifications from the application as I got a 11% notification instead of a 0% notification I was expecting, etc.

Clon1998, 2023-07-23:
> Do we know if the changes fixes an issue with notifications? I never got an android apk to test the earlier change.
>
>
>
> [image]
>
>
>
> I just started printing this file and the printer is still warming [...]

The notifications are handled completely by the companion. Since that is also using the old method i described in a previous reply it will still show the current layer based on your toolhead's position. 
Regarding the progress, mobileraker is using the file position to determine the progress. Since most Gcode files contain a ton of lines that contain comments regarding the slicer settings and also stuff like the preview image and start macro it can happen that the progress directly jumps to an higher number especially for small files.

Clon1998, 2023-07-23:
@fuzelet I apologize for the delay. Here is the APK file, which includes some additional development components that may be helpful.

https://github.com/Clon1998/mobileraker/releases/tag/pre-2.5.0

fuzelet, 2023-07-24:
> @fuzelet I apologize for the delay. Here is the APK file, which includes some additional development components that may be helpful.
> 
> https://github.com/Clon1998/mobileraker/releases/tag/pre-2.5.0

Thanks. I get an error when trying to install this though.

Pixel 6, Android 13 July Update

[image]

Clon1998, 2023-07-24:
> Thanks. I get an error when trying to install this though.
>
>
>
> Pixel 6, Android 13 July Update
>
>
>
>
>
> [image]

Probably the pixel is not allowing non signed side loaded apps. Sadly my development Android device broke so I have no way of testing it besides emulators

fuzelet, 2023-07-24:
> 

My phone is rooted and I got a prompt to allow unknown sources, just ended up with the error at the end. I'll see if there is anything else I can try on my phone. Appreciate the attempt. Have you considered a beta program in the play store that may alleviate this issue? Not sure the requirements for that just know they exist and wanted to mention it.

Clon1998, 2023-07-24:
> My phone is rooted and I got a prompt to allow unknown sources, just ended up with the error at the end. I'll see if there is anything else I can try on my phone. Appreciate the attempt. Have you [...]

I did, however the beta program on the play store works much different than apple and always requires me to collect emails for people to test.

Clon1998, 2023-08-08:
Hey there 👋,
I thought I will give you a small update regarding the accuracy of the progress, layers and eta.
I did have some time during the weekend and changed the implementation once again. Now, the progress, eta, current layer and max layer are matched with the implementation of Mainsail and thefore should not be different anymore. I also shortly tested that with the means available to me.
I also implemented these changes to the companion. Thefore, if you want to test it I invite you to try the new version of the companion available under the branch ‘rewrite‘

fuzelet, 2023-08-08:
Btw, I found the issue with the android application not installing. I had to uninstall the application first and then install the preview apk. I am on 2.5.0-145 now.

https://github.com/Clon1998/mobileraker/releases/download/pre-2.5.0/preview_b_2_5_0.apk

Clon1998, 2023-08-11:
Added in [2.5.0](https://github.com/Clon1998/mobileraker/blob/master/docs/changelog.md#250---2023-08-11)

closed 2023-08-11 by Clon1998