Clon1998/mobileraker #141: iOS Widget

dandrzejewski, 2023-04-07:
**Describe the solution you'd like**
I would like an iOS widget that contains information about my printer and the current print.  Progress percentage, ETA, and perhaps a frame from the camera.

thatbluflame, 2023-05-08:
Agreed, or even the new live notifications like octopod has, just switched over from octopod and I already miss them.

Clon1998, 2023-05-08:
> Agreed, or even the new live notifications like octopod has, just switched over from octopod and I already miss them.

I agree, but since I am using flutter for cross platform compatibility I rely on a 3rd party plugin for notifications.
Afaik, the dev is working on support for live notifications the last time i chatted with him.
But there is only so much I can do  as I am the only one actively developing mobileraker in my spare time😉

thatbluflame, 2023-05-08:
> I agree, but since I am using flutter for cross platform compatibility I rely on a 3rd party plugin for notifications. Afaik, the dev is working on support for live notifications the last time i [...]

Ah ok, thank you, app is still fantastic even without it!

hauserkristof, 2023-07-25:
Hi @Clon1998 !

Do you have any update on this?

Thanks,
Kristóf

Clon1998, 2023-07-25:
> Hi @Clon1998 !
> 
> Do you have any update on this?
> 
> Thanks, Kristóf

All I can update you on is that I added widgets for Android and Ios to my to-do list and also found some potential packages that can help me create them.