Clon1998/mobileraker #144: Error while trying to fetch printer

Edvinkajen, 2023-04-15:
This is the error code i get on the dashboard:

Exception:
 MobilerakerException{Missing field, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:71)
#1      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)
<asynchronous suspension>

All the other features is working such as notifications, console, files e.t.c. The only thing affedcted by this error is the dashboard.

 - Device: Iphone 12
 - OS: IOS 16.3.1

**Mobileraker-Version
2.2.2-146

Peaches491, 2023-04-15:
Same for me this morning

```
Exception:
 MobilerakerException{Missing field, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:71)
#1      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)
<asynchronous suspension>
```

Peaches491, 2023-04-15:
Also, I don’t know how you’re handling your logging setup, but it appears you have something like `--config=always` set, and that’s where all of the bracket number junk comes from at the beginning and end of each line (they’re bash color escape codes)

You should only use those escape codes in contexts where they’ll be interpreted (I.e in a TTY terminal) and not in this part of the app

Edvinkajen, 2023-04-15:
i got my app to wok now. It was a problem with my klipper config file. I had left out some gcode macros from my configuration files which caused this error. display_status, pause_resume, cancel_print, resume, pause. All these was left out of my config.

Peaches491, 2023-04-15:
Oh great! It sounds like there are some undocumented required macros, and that the error isn't being caught and handled properly. 

@Edvinkajen, care to share the macros you added?

Peaches491, 2023-04-15:
Eureka! I figured out the error:

I was missing a critical step in the mainsail setup:
https://github.com/mainsail-crew/mainsail-config#how-to-setup

I simply needed to add the following to my `printer.cfg` and everything started working!
```
[include mainsail.cfg]
```

Clon1998, 2023-04-15:
> Also, I don’t know how you’re handling your logging setup, but it appears you have something like `--config=always` set, and that’s where all of the bracket number junk comes from at the beginning [...]

Thanks for the hint. In fact that problem is related to IOS and my logging lib. I might look into customizing my logger setup, but for now, I don't care about these chars.

Clon1998, 2023-04-15:
> 

Awesome!
With that info I can reproduce the issue and improve the error handling in that cases ;)
I assume I can close the issue now?

Edvinkajen, 2023-04-17:
Yes you can close this issue 👍

closed 2023-04-17 by Clon1998