Clon1998/mobileraker #146: Octoeverywhere with Moonraker - API key

dushanL, 2023-04-19:
I have linked octoeverywhere to mobileraker. When mobileraker is out of local network, i receive warning message: "It seems like you configured trusted clients for Moonraker. Please add the API key in the printers setting!"
In moonraker.conf is force_logins: True, on local network work fine.  

If i disable force_logins octoeverywhere link work fine out of local network.

moonraker - v0.8.0-32-g31e589a
klipper - v0.11.0-173-ga8b1b0ef
octoeverywhere - 2.5.2-0-ge083fd4
mobileraker android version - 2.2.3
Device: Samsung Galaxy S22
OS: Android 13

Clon1998, 2023-04-19:
Hey, afaik that's exactly how the force login is supposed to work. However, you might wanna check with the ocotoeverywhere team to verify that. 
Did you try to add the API key to mobileraker? It should resolve this issue.

dushanL, 2023-04-19:
Yes, I have an API key in mobileraker.  With this setting, force login works for me only on the local network. 
[image]

I have an error via octoeverywhere.
[image]

Clon1998, 2023-04-19:
I see,
I will look into that.

mattconn24, 2023-04-30:
I can confirm that I am also having this same issue.

Clon1998, 2023-04-30:
> I can confirm that I am also having this same issue on my iPhone 14.

I am in contact with the author of OE. We are currently investigating.

Clon1998, 2023-04-30:
Short update:
I have found the potential problem on my side that prevented the usage of the API Key while using OE.
At the same time, the author of OE also figured out that on his side the implementation is/was missing.
I hope we can resolve that in the next week.

mattconn24, 2023-05-01:
Get news!  Thank you for the amazing work on this app.

Chris-Epler, 2023-05-09:
Wondering if there is an update on this and while troubleshooting it I noticed when I tried to scan the QR code from the web interface of Moonraker for the API key that it wasn't updating it and I had to type in the API key manually...

Clon1998, 2023-05-09:
> Wondering if there is an update on this and while troubleshooting it I noticed when I tried to scan the QR code from the web interface of Moonraker for the API key that it wasn't updating it and I had to type in the API key manually...

Are you talking about the QR not getting inserted into the correct field in the app?
If that is the case, this is not related

Clon1998, 2023-05-09:
I just triggered the build for Mobileraker v2.2.6. According to my testing, it should resolve any missing implementation regarding the API key for Octoeverywhere.
After chatting with the OE dev, he added full support for authed sockets in the OE client version 2.5.5. 
If you guys have any problem regarding Mobileraker+Oe with the API Key, please comment on this issue and I will reopen it.

closed 2023-05-09 by Clon1998

Clon1998, 2023-05-19:
Added in [2.3.0](https://github.com/Clon1998/mobileraker/releases/tag/android2.3.0)