Clon1998/mobileraker #147: KeenDNS

DrPerryCoke, 2023-04-20:
Greetings! Is it possible to access a printer using [KeenDNS](https://help.keenetic.com/hc/en-us/articles/360000400919-KeenDNS-service)?

Clon1998, 2023-04-20:
Hey,
I don't know that service. However, by reading through it briefly I believe it will work. But I am not sure how it secures your printer's connection since it seems like it only creates a permanent DNS for your router. Therefore, I'd suggest you have a look at either tailscale or ocotoeverywhere as both of these solutions provide a secure tunnel to your printer.

Clon1998, 2023-04-20:
Well there is Ocotoeverywhere integration that does exactly that.
If you're in the same network as the printer, it the app will use the local one. If the local connection fails it will use ocotoeverywhere if you have it linked

DrPerryCoke, 2023-04-21:
It is impossible to pay for Octoeverywhere in Russia

closed 2023-04-21 by DrPerryCoke

Clon1998, 2023-04-21:
> It is impossible to pay for Octoeverywhere in Russia

I see.
In that case your setup is fine. Just be sure it is secured and not exposed to the public internet!