Clon1998/mobileraker #150: Mobileraker iOS app gcode thumbnails error

Hr-Ravn, 2023-04-30:
Having issues with getting gcode thumbnails to show in mobileraker iOS App.
Working fine on browser!

Getting following errors in the log in iOS app

What can i do about that?

`[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/Shape-Box_120-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/Shape-Box_120-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/sb_adxl_mount_generic_15.5mm_c_c-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/sb_adxl_mount_generic_15.5mm_c_c-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/Shape-Box_120-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/Shape-Box_120-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/sb_adxl_mount_generic_15.5mm_c_c-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/sb_adxl_mount_generic_15.5mm_c_c-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/Shape-Box_120-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/Shape-Box_120-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/sb_adxl_mount_generic_15.5mm_c_c-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/sb_adxl_mount_generic_15.5mm_c_c-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;208m⚠️ http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[[38;5;196m⛔ HttpException: Invalid statusCode: 404, uri = http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png[0m]
[... 112 lines cut ...]
[... cut ...]
[[38;5;12m💡 Updating Default Macros for "Pumpkin(ws://192.168.87.175/websocket)"![0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 Stop MJPEG[0m]
[[38;5;12m💡 _AdaptiveStreamManager DISPOSED[0m]
[[38;5;208m⚠️ -DISPOSED printerSelected[0m]
[[38;5;12m💡 Start MJPEG - targFps: 15 - http://192.168.87.175/webcam/?action=stream[0m]
[[38;5;12m💡 Getting meta for file: `ASA/V0/Frame/3510_Side_Skirt_Mid_x2.gcode`[0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 Stop MJPEG[0m]
[[38;5;12m💡 _AdaptiveStreamManager DISPOSED[0m]
[[38;5;208m⚠️ -DISPOSED printerSelected[0m] `

Hr-Ravn, 2023-04-30:
Don't know what was wrong, but a re-install of the app did solve the issue.

Clon1998, 2023-04-30:
> Don't know what was wrong, but a re-install of the app did solve the issue.

First of all, I am glad that the issue resolved itself.

Here is however my take on the reasoning:
It seems like you misconfigured the **Printer - Address**

Have a look at the url that I copied from your log

http://192.168.87.175///server/files/gcodes/PLA/.thumbs/laptop_stand_remake-400x400.png

Somehow it contains three **///**, therefore the resolving of the image failed. 
I will reopen your issue and check how the app construct these URLs to ensure this won't happen again!

closed 2023-05-19 by Clon1998