Clon1998/mobileraker #152: M117 time remaining flashing in android app, makes UI jump

devsfan1830, 2023-04-30:
Actually unsure if its a bug or caused by a klipper plugin. I am using KAMP and an auto z plugin. My start gcode includes a blank M117. I use superslicer. But duing a print the mobileraker app UI jumps up and down due to an M117 time remaining message that seems to blink in and out. 

Video of behavior: https://youtube.com/shorts/akeES8MQgcs?feature=share

If theres anything helpful, like gcode, macro snippets, or logs I can provide I am happy to do so.

Clon1998, 2023-04-30:
Is there any reason why you use M117 for the time remaining display?

Could you provide me the Gcode? I am kinda curious why the M117 appears and disappears.

devsfan1830, 2023-04-30:
Attached a current gcode that is showing this behavior. My custom start gcode is simply:
M104 S0 ; Stops PS/SS from sending temp waits separately
M140 S0
M117
PRINT_START BED=60 EXTRUDER=205

It was said in a KAMP setup tutorial that the M117 in the start gcode is ncessaru to actually get it to do an adaptive mesh. They said all it does was clear any messages on the screen. There is no other source of an M117 that I personally added.

Had to zip the gcode to get it under the 25MB limit.

[Cab Logo Removed v1-Ellis - Voron 45 Degree (0.4mm)-Polymaker Polylite PLA.zip](https://github.com/Clon1998/mobileraker/files/11362553/Cab.Logo.Removed.v1-Ellis.-.Voron.45.Degree.0.4mm.-Polymaker.Polylite.PLA.zip)

devsfan1830, 2023-04-30:
also, this is my start macro

[gcode_macro PRINT_START]
#   Use PRINT_START for the slicer starting script - please customise for your slicer of choice
gcode:
    G90                 ; Absolute coordinates.
    M83                 ; Relative extruder mode 
    BED_MESH_CLEAR
    M117 Heating bed
    M190 S{ params.BED }       ; Wait for bed to get to target temperature.
    M117 Preheating nozzle
    M109 S150                  ; Wait for nozzle to heat to 150C - soften filament, but no oozing.
    M107                       ; Print fan off
    #INSERT NOZZLE CLEAN MACRO HERE
    G32                        ; Home All, QGL
    BED_MESH_CALIBRATE
    CALIBRATE_Z #ADD NOZZLE CLEAN INTO CALIBRATE Z
    G1 Z50 F3000              
    M117 Heating nozzle
    M109 S{ params.EXTRUDER }  ; Wait for nozzle to get to target temperature.
    G92 E0
    VORON_PURGE
    M117 Printing

devsfan1830, 2023-04-30:
So, I just realized my superslicer config had this enabled
[image]

If this is the culprit AND entirely unnecessary for print progress in klipper/mainsail, then I shall disable it and we can close this ticket.

Clon1998, 2023-05-01:
> So, I just realized my superslicer config had this enabled
> 
> [image]
> 
> 
> 
> If this is the culprit AND entirely unnecessary for print progress in klipper/mainsail, then I shall disable it and we can close this ticket.

afaik You don't need it

closed 2023-05-01 by Clon1998