Clon1998/mobileraker #153: Number in tld makes url invalid

UmbraAtrox, 2023-05-09:
**Describe the bug**
When a number is added to tld it makes the url invalid.

**To Reproduce**
1. Add printer
2. Enter printer-url "abc.ab1"

**Expected behavior**
App attempts to connect anyway (possibly after prompt to ask the user if they are really really really sure they didn't typo)

**Mobileraker-Version
2.2.4-156

Clon1998, 2023-05-09:
Hey, 
As a temporary workaround, add the printer with the domain ending with e.g. ´.com´. Then go into the printer's setting page. There is no validation on that page on either the WS or the HTTP address input. Therefore, update the correct tld there for now.

UmbraAtrox, 2023-05-15:
> Hey, As a temporary workaround, add the printer with the domain ending with e.g. ´.com´. Then go into the printer's setting page. There is no validation on that page on either the WS or the HTTP address input. Therefore, update the correct tld there for now.

There is no check on the printer-address but websocket-address does check, says 'Field requires a valid URL address' and doesn't let me save changes.

lifer0se, 2023-07-30:
Experiencing the same issue with mainsail. 
I'm trying to add it using the host name, but "http://mainsail" does not pass validation on "add printer".
I can add the printer using its IP address and then edit the printer address, but the websocket address is still giving me trouble.

Any workarounds to not using a static IP? Could you maybe disable validation entirely?

Clon1998, 2023-07-30:
> Experiencing the same issue with mainsail.
>
> I'm trying to add it using the host name, but "http://mainsail" does not pass validation on "add printer".
>
> I can add the printer using its IP address [...]

The next release will offer a new add printer flow that allows to completely disabled validation.

If you are using iOS i can send you a TestFlight link. Otherwise you need to wait until it is ready.

lifer0se, 2023-07-30:
> The next release will offer a new add printer flow that allows to completely disabled validation.
> 
> If you are using iOS i can send you a TestFlight link. Otherwise you need to wait until it is ready.

Great to hear! it's not that big an issue for me right now, I just have to change the IP if router resets.

I am on iOS, so I'd like a TestFlight invite, I'm decent at beta testing :)

Clon1998, 2023-07-30:
Here you go

https://testflight.apple.com/join/h9JwuyR0

Clon1998, 2023-08-11:
Fixed in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0)

closed 2023-08-11 by Clon1998