Clon1998/mobileraker #158: Dashboard error, other tabs seems to be working

etet100, 2023-05-15:
**Describe the bug**
I have the latest Mobileraker and the latest Moonraker. The dashboard is generating an error. Other tabs (console, files) seem to be working just fine. I'm using a pretty old mobile with Android 8. 

```
Exception:
 MobilerakerException{Missing field, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:71)
#1      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)
<asynchronous suspension>
```

**To Reproduce**
Configure printer, connect, go to dashboard

**Expected behavior**
Dashboard should appear

**Smartphone (please complete the following information):**
Device: Blackberry Keyone
Android: 8

**Mobileraker-Version
2.2.4-156

Clon1998, 2023-05-15:
> 2.2.6 freezes my mobile. I see loading screen (orange icon) and i can't even kill it.
> 
> 2.2.5 shows exactly the same error as 2.2.4. And it generally is working slow.

2.2.5 should show a bit more info regarding the missing field. Just post the entire error please.

etet100, 2023-05-15:
Version: 2.2.5-145
```
Exception:
 MobilerakerException{Missing field: printFan, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:67)
#1      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)
<asynchronous suspension>
```
And debug log
```
[[38;5;12m💡 Selecting machine My Printer[0m]
[[38;5;12m💡 Started initializeAvailableMachines[0m]
[[38;5;196m⛔ machineProvider creation STARTED 0911fd94-1378-469a-b165-333e40bfc364[0m]
[[38;5;196m⛔ machineProvider creation DONE 0911fd94-1378-469a-b165-333e40bfc364[0m]
[[38;5;12m💡 initialized all machineProviders[0m]
[[38;5;12m💡 Init for My Printer(0911fd94-1378-469a-b165-333e40bfc364)[0m]
[[38;5;12m💡 [654528446]Trying to connect to ws://192.168.1.25:7125/websocket[0m]
[[38;5;12m💡 [654528446-ws://192.168.1.25:7125/websocket] ClientState.disconnected ➝ ClientState.connecting[0m]
[[38;5;12m💡 No remote client configured... cant do handover! ref:62081831[0m]
[[38;5;12m💡 Finished initializeAvailableMachines[0m]
[[38;5;196m⛔ ääää AsyncData<ClientState>(value: ClientState.disconnected)[0m]
[[38;5;12m💡 Notifications are permitted: true[0m]
[[38;5;12m💡 Token from FCM eL1uEPEOTmCPCZjZKWYwdz:APA91bFGirFPIMka7oPh2Qoai6gf1y4wTmAGYuhgZ2JcOHVNqZnlc61Sn-Hwj2d_kM7agDAJLVGKKjWNN0dFESYl8_hda-YgAMlnT4xJtFN1rBlGfUPDxEh9V-IytRR7zG34Xq89B4fF[0m]
[[38;5;12m💡 Device's FCM token: eL1uEPEOTmCPCZjZKWYwdz:APA91bFGirFPIMka7oPh2Qoai6gf1y4wTmAGYuhgZ2JcOHVNqZnlc61Sn-Hwj2d_kM7agDAJLVGKKjWNN0dFESYl8_hda-YgAMlnT4xJtFN1rBlGfUPDxEh9V-IytRR7zG34Xq89B4fF[0m]
[[38;5;12m💡 [478672159]Trying to connect to ws://192.168.1.25:7125/websocket[0m]
[[38;5;12m💡 [478672159-ws://192.168.1.25:7125/websocket] ClientState.disconnected ➝ ClientState.connecting[0m]
[[38;5;12m💡 No remote client configured... cant do handover! ref:764311301[0m]
[[38;5;196m⛔ ääää AsyncData<ClientState>(value: ClientState.disconnected)[0m]
[[38;5;12m💡 Token from FCM eL1uEPEOTmCPCZjZKWYwdz:APA91bFGirFPIMka7oPh2Qoai6gf1y4wTmAGYuhgZ2JcOHVNqZnlc61Sn-Hwj2d_kM7agDAJLVGKKjWNN0dFESYl8_hda-YgAMlnT4xJtFN1rBlGfUPDxEh9V-IytRR7zG34Xq89B4fF[0m]
[[38;5;12m💡 [478672159-ws://192.168.1.25:7125/websocket] ClientState.connecting ➝ ClientState.connected[0m]
[[38;5;12m💡 >>>Fetching Server.Info[0m]
[[38;5;12m💡 >>>Fetching Printer.Info[0m]
[[38;5;12m💡 <<<Received Server.Info[0m]
[[38;5;12m💡 <<<Received Printer.Info[0m]
[[38;5;12m💡 Refreshing printer for uuid: 0911fd94-1378-469a-b165-333e40bfc364[0m]
[[38;5;12m💡 >>>Querying printers object list[0m]
[[38;5;12m💡 Jrpc Client of My Printer(ws://192.168.1.25:7125/websocket) is connected, can Setup FCM on printer now![0m]
[[38;5;12m💡 Getting fcm.0911fd94-1378-469a-b165-333e40bfc364[0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;12m💡 <<<Received printer objects list![0m]
[[38;5;12m💡 >>>Querying Printer Objects![0m]
[[38;5;12m💡 Current FCMConfig in MoonrakerDB: DeviceFcmSettings{machineId: eL1uEPEOTmCPCZjZKWYwdz:APA91bFGirFPIMka7oPh2Qoai6gf1y4wTmAGYuhgZ2JcOHVNqZnlc61Sn-Hwj2d_kM7agDAJLVGKKjWNN0dFESYl8_hda-YgAMlnT4xJtFN1rBlGfUPDxEh9V-IytRR7zG34Xq89B4fF, machineName: My Printer, language: en, settings: NotificationSettings{progress: 0.25, states: {standby, printing, paused, complete, error}}}[0m]
[[38;5;12m💡 <<<Received queried printer objects[0m]
[[38;5;12m💡 Fetching cached temperature store data[0m]
[[38;5;12m💡 Received cached temperature store for [extruder][0m]
[[38;5;196m[0m[39m[48;5;196mMobilerakerException{Missing field: printFan, parentException: null, parentStack: null}[0m[49m, [38;5;196m#0   PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:67)[0m, [38;5;196m#1   PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)[0m, [38;5;196m#2   <asynchronous suspension>[0m, [38;5;196m⛔ Unexpected exception thrown during refresh 0911fd94-1378-469a-b165-333e40bfc364...[0m]
[[38;5;196m[0m[39m[48;5;196mMobilerakerException{Missing field: printFan, parentException: null, parentStack: null}[0m[49m, [38;5;196m#0   PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:67)[0m, [38;5;196m#1   PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)[0m, [38;5;196m#2   <asynchronous suspension>[0m, [38;5;196m⛔ Error in Dash[0m]
[[38;5;12m💡 Fetching cached GCode commands[0m]
[[38;5;12m💡 Fetching available GCode commands[0m]
[[38;5;12m💡 Received cached GCode commands[0m]
[[38;5;12m💡 Received 54 available GCode commands[0m]
[[38;5;12m💡 Fetching for `gcodes` [extended:true][0m]
[[38;5;196m[0m[39m[48;5;196mMobilerakerException{Missing field: printFan, parentException: null, parentStack: null}[0m[49m, [38;5;196m#0   PrinterBuilder.build (package:mobileraker/data/dto/machine/printer.dart:67)[0m, [38;5;196m#1   PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)[0m, [38;5;196m#2   <asynchronous suspension>[0m, [38;5;196m⛔ Error in Dash[0m]
[[38;5;12m💡 App in AppLifecycleState.inactive[0m]
[[38;5;12m💡 App backgrounded[0m]
[[38;5;12m💡 App forgrounded[0m]
[[38;5;12m💡 Refreshing selectedPrinter...[0m]
[[38;5;12m💡 No remote client configured... cant do handover! ref:764311301[0m]
[[38;5;12m💡 App in AppLifecycleState.inactive[0m]
[[38;5;12m💡 App backgrounded[0m]
[... cut ...] [0m, [38;5;196m#1   PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:204)[0m, [38;5;196m#2   <asynchronous suspension>[0m, [38;5;196m⛔ Error in Dash[0m]
[[38;5;12m💡 App in AppLifecycleState.inactive[0m]
[[38;5;12m💡 App backgrounded[0m]
[[38;5;12m💡 App forgrounded[0m]
[[38;5;12m💡 Refreshing selectedPrinter...[0m]
[[38;5;12m💡 No remote client configured... cant do handover! ref:764311301[0m]
[[38;5;208m⚠️ -DISPOSED printerSelected[0m]
```

Clon1998, 2023-05-15:
Thanks!
It seems like you're running your printer without a parts fan. At least, that's why mobileraker is complaining.
I just recently learned that Klipper does not enforce/require the configuration of a print fan.
I will leave this issue open, and rename it to make sure I don't forget to refactor my implementation to allow for printers without print fan.

A temporary solution for this might be to configure a print fan config with an unused pin of your control board.

etet100, 2023-05-15:
Thanks. My part fan is connected directly to 5V. It's always on.

Clon1998, 2023-05-19:
Added in [2.3.0](https://github.com/Clon1998/mobileraker/releases/tag/android2.3.0)

closed 2023-05-19 by Clon1998