Clon1998/mobileraker #164: Webcams not working with latest update

spitfire1337, 2023-05-20:
After the latest may 19th update, my webcams have stopped working for my 1 printer that has multiple webcams. 

An error is in its place stating: "type null is not a subtype of map<string,dynamic>" with a refresh button below it that doesn't work. Attempting to go into the settings for the printer to add the webcams back and it displays the same error where the webcams would be listed. Selecting the add button also doesn't work.

Webcams for my other printers with just 1 camera are still working fine.

Webcams work in fluidd fine and we're working in mobileraker up until the update got installed. Phone is a Samsung Galaxy s21 ultra.
[image]
[image]

Clon1998, 2023-05-20:
Hey,
thanks for reporting this issue. This might be related to an issue with syncing the cams from fluidd/mainsail with mobileraker. I think I have resolved this issue with 38b6fac741b950657ad2d007fe56373c345927d8.
I will trigger a Hotfix today that should hopefully be available by tomorrow

spitfire1337, 2023-05-21:
Perfect, I just got the update and was able to re-add my webcams back. Thanks for the quick response & quick fix!

closed 2023-05-21 by spitfire1337