Clon1998/mobileraker #165: Stuck on logo screen

ishtangli, 2023-05-21:
Pixel 6 pro on android 13.

Just shows the logo on latest update.

Clon1998, 2023-05-21:
Hey,
sadly I am unable to reproduce this issue with any scenario I can currently think of.
Therefore, can you please try to remove the app's locale data and retry starting the app?

ishtangli, 2023-05-21:
How can I do that? Tried uninstalling, reinstalling, deleting cache.

ishtangli, 2023-05-21:
Finally got it working by deleting the cache but NOT uninstalling the app.

Clon1998, 2023-05-21:
> Finally got it working by deleting the cache but NOT uninstalling the app.

Hey,
sorry didn't get a notification earlier. I am glad it's working now for you!
Androids uninstall might not delete any of the saved files. Thefore clearing the cache worked.

I will look into that!

closed 2023-06-28 by Clon1998