Clon1998/mobileraker #168: Webcam Error

frankt2012, 2023-05-22:
**Describe the bug**
Webcam no longer working after update

**To Reproduce**
Try to view webcam

**Expected behavior**
Webcam shows when starting app

**Smartphone (please complete the following information):**
 - Device: Pixel 7 Pro
 - OS: Android 13

**Mobileraker-Version
2.3.1-172

**Additional context**
When trying to view the Webcam I get HttpException: Strem returned 404 status. Everything was working last I checked just fine. Also using this with Sonic Pad. Don't know if that matters.

Clon1998, 2023-05-22:
Hey,
thanks for reporting this.
In version 2.3.0 I migrated from local webcam setting storage to syncing/using the same as fluidd/mainsail. Thefore can you check the address that is shown in the webcam settings?
I have the strong suspicion that the address uses the local host address (127.0.0.1 or localhost).

frankt2012, 2023-05-22:
> Hey, thanks for reporting this. In version 2.3.0 I migrated from local webcam setting storage to syncing/using the same as fluidd/mainsail. Thefore can you check the address that is shown in the [...]

I show this when I click on the camera feed in fluid http://192.168.1.209/webcam/?action=stream&cacheBust=1684768098898.
When I look under camera settings in fluid it dosen't show me much. Here is a link to a picture of the screen I took of the camera settings https://photos.app.goo.gl/koRGCfcRx6aGGGPa9.

Clon1998, 2023-05-22:
Alright, what is shown in the webcam settings of Mobileraker?

To access these:
1. Open sidebar by clicking the top left three bar icon. The side menu should open
2. At the top of the sidebar should be the name of your printer and right next to it a setting icon. Click on that. (The printer settings page should open now)
3. Scroll down to the webcam settings
4. Click on the correct webcam to expand its settings.

Clon1998, 2023-05-22:
That looks more than fine. Can you post a screenshot of the content of the webcam settings?
In addition, can you also post a screenshot of the printer's websocket+http URL?

frankt2012, 2023-05-23:
> That looks more than fine. Can you post a screenshot of the content of the webcam settings? In addition, can you also post a screenshot of the printer's websocket+http URL?

[image]
[image]
[image]

Hers some pictures of all the setting. Everything is exactly as I had it when it was working. I even tried completely reinstalling the Mobileraker app last night and putting the printer back. The only thing I can think of is something changed on the Sonic Pad end during the last update. I can't remember if I used the app since then.

Clon1998, 2023-05-23:
Gotcha.
I think the problem is caused by the port number you included. I will adjust my implementation to ensure it removes the port number for any relative parts.
In the meantime to resolve this issue for you, you can use this address as webcam address:

http://192.168.1.209/webcam/? action=stream

frankt2012, 2023-05-23:
> Gotcha. I think the problem is caused by the port number you included. I will adjust my implementation to ensure it removes the port number for any relative parts. In the meantime to resolve this issue for you, you can use this address as webcam address:
> 
> http://192.168.1.209/webcam/?action=stream

Ok thank you for the help. I will try that when I get home. I have to the port number on because of the way the Sonic Pad works . You can have four printers hooked at once so I guess they use the port numbers to differentiate between them.

Clon1998, 2023-05-23:
> Ok thank you for the help. I will try that when I get home. I have to the port number on because of the way the Sonic Pad works . You can have four printers hooked at once so I guess they use the [...]

No problem. As I said, it's just an issue in my implementation that I do not remove the port number from the printer address when I combine the relative webcam address with the printer address.
As I said the current solution for this is to simply provide a absolute path to the webcam. In that case my app does not do any merging!

frankt2012, 2023-06-02:
> No problem. As I said, it's just an issue in my implementation that I do not remove the port number from the printer address when I combine the relative webcam address with the printer address. As I [...]

I just finally got around to trying the fix you said and it worked. Thank you for the help.

Clon1998, 2023-06-02:
> I just finally got around to trying the fix you said and it worked. Thank you for the help.

Awesome! I already fixed this issue for the upcoming release but you can keep the workaround after upgrading the app as well!

closed 2023-06-20 by Clon1998