Clon1998/mobileraker #169: Dev Contact? Missing Feature?

Galapagon, 2023-05-24:
I'm not sure if I'm missing where to select this in the app, or if it's not a current feature. Additionally, I wasn't sure how to reach out to ask, without opening an issue.

I'm trying to find "bed screws adjust" as I would with Mainsail desktop. This automatically goes around the screw points on the heat bed and asks if I adjusted the screw or not. Helps with bed leveling.

Is this available in Mobileraker?

Thank you for your time!

Clon1998, 2023-05-25:
Hey,
GitHub issue is fine. However, on the supporter page, the app should offer a Dialog with direct contact details to me.
I don't own a Printer that can make use of the `BED_SCREWS_ADJUST ` command hence it is missing in Mobileraker. Therefore, do you mind sharing an example video that showcases the usage of this command in the UI?

Galapagon, 2023-05-25:
Thank you for checking, if I can assist more please let me know!

https://imgur.com/a/FJcufWc

From printer.cfg:

[screws_tilt_adjust]
screw1: 23,63
screw1_name: front left
screw2: 143,63
screw2_name: front right
screw3: 143,183
screw3_name: back right
screw4: 23,183
screw4_name: back left
speed: 100.0
screw_thread: CW-M5

closed 2023-06-20 by Clon1998