Clon1998/mobileraker #173: Support for Job queue

luisillo26, 2023-06-01:
It would be great you could implement support for Job Queue, I mean the ability to add gcode files to the job queue, start printing, reorder, remove and set batch quantity to the Job Queue.

A clear and concise description of what you want to happen.

This feature is implemented in Mainsail and I believe is part of the moonraker api.

[image]

[image]

Clon1998, 2023-08-20:
Here is a small preview of the JobQueue. I would love to get some feedback on it (Ignore some of the placeholders)

[image]

[image]

bitingsock, 2023-09-20:
Is this included in 2.6.3? will this add support for uploading gcode?

Clon1998, 2023-09-20:
ohh yeah I am sorry missed that issue.
It was added in 2.6.2

closed 2023-09-20 by Clon1998