Clon1998/mobileraker #175: Dialog/Wizard for screws_tilt_adjust

Clon1998, 2023-06-09:
According to the information provided in the documentation of Klipper, specifically in the section titled **_Adjusting Bed Leveling Screws Using the Bed Probe_** (accessible at: [Klipper docs](https://www.klipper3d.org/Manual_Level.html#adjusting-bed-leveling-screws-using-the-bed-probe)), the `SCREWS_TILT_CALCULATE` command serves to activate an automated method for determining the necessary adjustments to be made on the bed screws in order to achieve a flat print surface. Presently, Mobileraker incorporates a feature that allows the user to trigger this command through a button; however, it lacks a configuration wizard designed to assist the user in this process.

To address this limitation, it is recommended to introduce a dialogue box similar to the existing `BED_SCREWS_ADJUST` dialogue. This new dialogue box would provide the user with comprehensive information regarding the current screw configuration and the specific adjustments required for each screw. By presenting this information in a clear and organized manner, users will be better equipped to carry out the necessary adjustments effectively.

jamincollins, 2024-04-03:
Would love to see this added.

Clon1998, 2024-04-18:
Added this via [8545be8](https://github.com/Clon1998/mobileraker/commit/8545be8c26bb4204c0d7df9c1dd89976b6ae2a07).

Please let me know if something is missing!

[image]

commit 8545be8c (2024-04-18): Resolve #175 - Added screws_tilt_adjust_dialog