Clon1998/mobileraker #180: mobileraker can not run.FormatException: Unexpected  character (at line 23, character 6)

ss8688, 2023-06-25:
I cannot run the app.
.FormatException: Unexpected  character (at line 23, character 6)

[image]

 - Device: redmi k60 pro
 - OS: miui14.0.23 (Android13)

 -Mobileraker2.4.0

ealin1994, 2023-06-25:
[image]I have the same problem！

msoushi, 2023-06-25:
Same here with version 2.4.0 on Samsung Note 20 Ultra with OneUI 5.1 (Android 13).

hany, 2023-06-25:
I'm running on iOS. The app simply crashes immediately when trying to open the app. There is no visible error message, but I assume it's related to this issue.

Clon1998, 2023-06-25:
Hey, I will look into this. I just published 2.4.1 that should allow me to look into the error.

Clon1998, 2023-06-25:
Ohh nvm. I assume all of you are running the app in Chinese?

hany, 2023-06-25:
I am running in English on iOS. I assumed it was related as it just started happening with 2.4.0. Perhaps my issue is different. The app simply crashes immediately. Should I open a new issue?

msoushi, 2023-06-25:
Just tried 2.4.1 on my Android, looks like the problem still persists. It showed a slightly different error message though:
`FormatException: Unexpected character (at line 23, character 9)`

Clon1998, 2023-06-25:
> I am running in English on iOS. I assumed it was related as it just started happening with 2.4.0. Perhaps my issue is different. The app simply crashes immediately. Should I open a new issue?

Yes your issue might be related to something else. 
I could you try this TestFlight version?
Worst case I get the actual error report to look into:

https://testflight.apple.com/join/h9JwuyR0

Clon1998, 2023-06-25:
I just pushed 2.4.1-197, which resolves the issue for Chinese users :)

closed 2023-06-25 by Clon1998

maydayVFR, 2023-06-27:
I have a crash log for you as well if needed. Running in English on 14.1. Looks like an iOS 14.3 vs ios 16 build issue. So I could be out of luck. 
Can upload full crash log at request. 
```

Termination Description: DYLD, Symbol not found: _objc_claimAutoreleasedReturnValue | Referenced from: /private/var/containers/Bundle/Application/880C0DD4-A5FC-4188-A814-EF042B1601EA/Runner.app/Frameworks/flutter_native_splash.framework/flutter_native_splash (which was built for iOS 16.4) | Expected in: /usr/lib/libobjc.A.dylib

```