Clon1998/mobileraker #181: Refresh Printer Error

renfordt, 2023-06-25:
**Describe the bug**
App can't connect/refresh printer

**To Reproduce**
Steps to reproduce the behavior:
1. Open App
2. Error page is shown

**Expected behavior**
App connects to the printer without errors. 

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: Google Pixel 4a
 - OS: Android 13
 - Browser: Various

**Mobileraker-Version
2.4.1

**Additional context**
If more info is needed let me know.

Clon1998, 2023-06-25:
Hey,
thanks for your Bug report.
I have already noticed that issue and fixed it in 03c3aca53ab3560006e29ee67577697a75b45bba, will be included in a fix later this week.
I am not 100% sure when Klipper goes into the `cancelled` state, but as soon as your printer is back to ready (Restart Klipper), or printing the app should work as expected.

renfordt, 2023-06-27:
Thank you for your feedback.
In my case Klipper got int the cancelled state due to a failed END_PRINT Macro (Retract value higher than limit). Also it should be in the cancelled state when you manually cancel the print.

Clon1998, 2023-06-27:
> Thank you for your feedback.
> 
> In my case Klipper got int the cancelled state due to a failed END_PRINT Macro (Retract value higher than limit). Also it should be in the cancelled state when you manually cancel the print.

Out of curiosity I checked when Klipper added this state. It was added more than a year ago and yet, I never encountered it and you are the first user reporting the problem.

DrFate09, 2023-06-27:
@Clon1998 I believe I've also been seeing the same error. 

[image]
[image]

Clon1998, 2023-06-27:
> @Clon1998 I believe I've also been seeing the same error.
> 
> [image] [image]

Hey,
no yours is different.

Does this happen all the time for you or just in a particular case?

Clon1998, 2023-06-27:
Do you happen to have a `virtual_sdcard ` section in your config?

DrFate09, 2023-06-27:
@Clon1998 I created a new issue 👍

closed 2023-06-28 by Clon1998