Clon1998/mobileraker #183: App crashes immediately on startup

hany, 2023-06-26:
**Describe the bug**
On iOS, the app crashes immediately on startup. Installed the TestFlight version as requested in https://github.com/Clon1998/mobileraker/issues/180#issuecomment-1606089983 and submitted the crash report. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open app on iOS

**Expected behavior**
App should start up normally. 

**Smartphone (please complete the following information):**
 - Device: iPhone 7
 - OS: iOS 15.7.7

**Mobileraker-Version
2.4.0 and 2.4.1 TestFlight version

Clon1998, 2023-06-26:
Hey, the latest version should be available now on the stores with a fix for this issue.

hany, 2023-06-26:
Hey there, I see that 2.4.1 is out in the App Store, however the app still crashes immediately. Is there a newer version than 2.4.1?

Clon1998, 2023-06-26:
Ohh you're issue was not related to the Chinese translation was it?

hany, 2023-06-26:
It was not. I'm running it in English. I submitted the crash report via TestFlight as requested.

Clon1998, 2023-06-26:
Right, let's see if I receive some error logs from either firebase or via TestFlight.
For me and others the latest version works just fine.
You can also try to entirely remove the app and reinstalled it to ensure its a fresh install.

hany, 2023-06-26:
Tried removing the app and data entirely and re-downloaded. Issue still occurs. 

I had sent the crash report via TestFlight. Did you receive the report yet? Should I try again?

Clon1998, 2023-06-26:
I am looking into it

JMSPI, 2023-06-26:
I'm having the same issue on an iPhone XS on IOS 15.3. I have submitted a crash report through TestFlight.

JMSPI, 2023-06-27:
I updated to IOS 16.5.1 and the app now starts up normally. Seems like the issue is caused by older IOS versions.

Clon1998, 2023-06-27:
I am currently investigating the issue!

Clon1998, 2023-06-27:
I will close this issue in favor of #186.
Please see my comment here https://github.com/Clon1998/mobileraker/issues/186#issuecomment-1610113978

closed 2023-06-27 by Clon1998