Clon1998/mobileraker #186: Immediate crash iOS 14.1.

maydayVFR, 2023-06-27:
Just updated to latest version today. Was also crashing yesterday but yesterday was my first usage of Mobileraker so previous successful builds n/a. Running in English on 14.1. Looks like an iOS 14.3 vs ios 16 build issue. So I could be out of luck. 
Can upload full crash log at request. 

```

Termination Description: DYLD, Symbol not found: _objc_claimAutoreleasedReturnValue | Referenced from: /private/var/containers/Bundle/Application/880C0DD4-A5FC-4188-A814-EF042B1601EA/Runner.app/Frameworks/flutter_native_splash.framework/flutter_native_splash (which was built for iOS 16.4) | Expected in: /usr/lib/libobjc.A.dylib

```

ripinteer, 2023-06-27:
Same issue here

DelukaDZ, 2023-06-27:
Same here on IOS 15.7.6

Clon1998, 2023-06-27:
Hey,
I am working on it!

Clon1998, 2023-06-27:
Related to #183

FastivProService, 2023-06-27:
Good day. It also does not work, immediately closes. iOS 15.7.6

Clon1998, 2023-06-27:
Hey,
I am still not sure why theses crashes are happening. However, I changed some dependencies and got the app working on:
- IOS16.4
- IOS15.0
- IOS14.1

I am currently building this version for testflight and will post an invitation link to it as soon as the build is ready.
Please note that this version has Notifications disabled due to another issue I was unable to resolve today.

In order to ensure an easier debugging process for me, please completely remove Mobileraker before installing the Testflight version!

**Testflight Link**: WIP (I Will update as soon as available.)

[image]

Clon1998, 2023-06-28:
Here ia the TestFlight invitation:

https://testflight.apple.com/join/kO7PuGUL

maydayVFR, 2023-06-28:
And we're back! Will test thoroughly if you need but the immediate crash appears fixed! Good work! Thank you!

maydayVFR, 2023-06-28:
Everything appears to be in working order on my end. Printer is there, responding, stats line up. Can wait for others to chime in to close or whatever you need.

Clon1998, 2023-06-28:
@maydayVFR thanks for your feedback and I am quite happy that it seems to be fixed. Let's wait for more users in this issue to give feedback just to be sure!

Feel free to test stuff and report any findings.
I just recently added Crashlyrics and discovered a several issues users never reported so far.

Clon1998, 2023-06-28:
@DelukaDZ , @ripinteer , @FastivProService. Can you guys give me feedback if the testflight version managed to resolve the issue for you too?

ripinteer, 2023-06-28:
My apologies for delayed answer, I have 2.4.1 version downloaded with testflight and it still crashes(
Am have the latest version? How can I help you?

Clon1998, 2023-06-28:
> My apologies for delayed answer, I have 2.4.1 version downloaded with testflight and it still crashes( Am I have the latest version? How can I help you?

Please try this invite, the new version should be 2.4.2
https://testflight.apple.com/join/kO7PuGUL

ripinteer, 2023-06-28:
[image]
[image]
Tried it a couple times, still shows 2.4.1

maydayVFR, 2023-06-28:
@Clon1998 will do. In the meantime I'll try to break it haha.

Clon1998, 2023-06-28:
> [image]
> 
> [image]
> 
> Tried it a couple times, still shows 2.4.1

You might be in a different test group, i will look into that tomorrow. You should get a notification once 2.4.2 is ready

ripinteer, 2023-06-29:
Updated, now it works as normal. Thank you
I’ll do a little testing now

closed 2023-07-01 by Clon1998