Clon1998/mobileraker #187: Refresh Printer Error, could not parse

DrFate09, 2023-06-27:
**Describe the bug**
Been seeing this error recently after running into some printing errors in Klipper (related to setting up Klicky macro). 

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: Android Pixel 6A

**Mobileraker-Version
2.4.1

DrFate09, 2023-06-27:
I do have it included in my config. 
[image]

Clon1998, 2023-06-27:
> I do have it included in my config. [image]

Mhh okay,
can you send me the output of when you visit this site:
`http://YOUR-PRINTER-IP/printer/objects/list` (make sure to replace YOUR-PRINTER-IP)

DrFate09, 2023-06-27:
{"result": {"objects": ["webhooks", "configfile", "mcu", "gcode_move", "print_stats", "virtual_sdcard", "pause_resume", "display_status", "gcode_macro CANCEL_PRINT", "gcode_macro PAUSE", "gcode_macro RESUME", "gcode_macro SET_PAUSE_NEXT_LAYER", "gcode_macro SET_PAUSE_AT_LAYER", "gcode_macro SET_PRINT_STATS_INFO", "gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL", "gcode_macro _CLIENT_EXTRUDE", "gcode_macro _CLIENT_RETRACT", "gcode_macro CLEAN_NOZZLE", "gcode_macro UNSAFE_LOWER_BED", "gcode_macro START_PRINT", "gcode_macro END_PRINT", "gcode_macro WIPE_LINE", "gcode_macro DISABLE_MOTORS", "gcode_macro M600", "gcode_macro LOAD_FILAMENT", "gcode_macro UNLOAD_FILAMENT", "gcode_macro TEST_SPEED", "gcode_macro GET_PROBE_LIMITS", "gcode_macro LAZY_HOME", "gcode_macro ADXL_SHAPE_ALL", "gcode_macro ADXL_SHAPE_X", "gcode_macro ADXL_SHAPE_Y", "gcode_button sensor_fs", "gcode_button sensor_fu", "gcode_macro filament_change_state1", "gcode_macro filament_change_state2", "gcode_macro filament_load", "gcode_macro filament_unload", "gcode_macro _User_Variables", "gcode_macro _Probe_Variables", "gcode_macro _klicky_check_variables_version", "gcode_macro _KlickyDebug", "gcode_macro _exit_point", "gcode_macro _entry_point", "gcode_macro _Homing_Variables", "gcode_macro Attach_Probe_Lock", "gcode_macro Dock_Probe_Unlock", "gcode_macro _Probe_Unlock", "gcode_macro _Probe_Lock", "gcode_macro _DeployKlickyDock", "gcode_macro _RetractKlickyDock", "gcode_macro Attach_Probe", "gcode_macro Dock_Probe", "gcode_macro PROBE_CALIBRATE", "gcode_macro PROBE_ACCURACY", "gcode_macro _Umbilical_Path", "gcode_macro _Home_Z", "gcode_macro _CheckProbe", "gcode_macro _SetProbeState", "gcode_macro _Park_Toolhead", "gcode_macro _klicky_status_ready", "gcode_macro _klicky_status_busy", "gcode_macro _klicky_status_leveling", "gcode_macro _klicky_status_homing", "gcode_macro _klicky_status_cleaning", "gcode_macro _klicky_status_meshing", "gcode_macro _klicky_status_calibrating_z", "gcode_macro BED_MESH_CALIBRATE", "gcode_macro SETUP_KAMP_MESHING", "gcode_macro LINE_PURGE", "gcode_macro SETUP_LINE_PURGE", "z_tilt", "bed_screws", "screws_tilt_adjust", "probe", "bed_mesh", "heaters", "heater_bed", "fan", "heater_fan hotend_fan", "stepper_enable", "controller_fan octopus_fan_1", "controller_fan octopus_fan_2", "controller_fan rpi_fan", "exclude_object", "temperature_host raspberry_pi", "temperature_sensor raspberry_pi", "temperature_sensor mcu_temp", "tmc2209 stepper_x", "tmc2209 stepper_y", "tmc2209 stepper_z", "tmc2209 stepper_z1", "tmc2209 extruder", "motion_report", "query_endstops", "idle_timeout", "system_stats", "manual_probe", "toolhead", "extruder"]}}

Clon1998, 2023-06-28:
> Been seeing this error recently after running into some printing errors in Klipper (related to setting up Klicky macro).

I overlooked that particular section and assumed that the app always displayed it. However, I realized that the error was caused by a misconfigured macro. Could you please provide me with the details of the misconfiguration so that I can reproduce the issue?

Clon1998, 2023-06-28:
Steps to reproduce:
1. Start app normally on Dashboard mode (Printer/Klipper is normal too)
2. Modify a config either via app or PC to force a klipper config error (E.g. remove a stepper section or comment out an include)
3. Dashboard shows the klipper error
4. Trigger a RefreshPrinter
5. Boom error

closed 2023-06-28 by Clon1998