Clon1998/mobileraker #189: Unable to fetch files

AniTexs, 2023-06-28:
**Describe the bug**
Unable to see files on creality pad

**To Reproduce**
Steps to reproduce the behavior:
1. Go to printer
2. Click on files
3. See error

**Expected behavior**
To view files

**Smartphone (please complete the following information):**
 - Device: iPhone 13
 - OS: 16.5.1

**Mobileraker-Version
2.4.1-197

**Additional context**
I have a creality pad.

Here is the error dump
```

Exception:
 type '_OneByteString' is not a subtype of type 'int'

#0      new GCodeThumbnail.fromJson (package:mobileraker/data/dto/files/gcode_thumbnail.dart:19)
#1      new GCodeFile.fromJson.<anonymous closure> (package:mobileraker/data/dto/files/gcode_file.dart:111)
#2      MappedIterable.elementAt (dart:_internal/iterable.dart:375)
#3      ListIterator.moveNext (dart:_internal/iterable.dart:344)
#4      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189)
#5      new _GrowableList.of (dart:core-patch/growable_array.dart:150)
#6      new List.of (dart:core-patch/array_patch.dart:52)
#7      SetMixin.toList (dart:collection/set.dart:118)
#8      new GCodeFile.fromJson (package:mobileraker/data/dto/files/gcode_file.dart:111)
#9      FileService._parseDirectory.<anonymous closure> (package:mobileraker/service/moonraker/file_service.dart:340)
#10     new _GrowableList.generate (dart:core-patch/growable_array.dart:136)
#11     FileService._parseDirectory (package:mobileraker/service/moonraker/file_service.dart:336)
#12     FileService.fetchDirectoryInfo (package:mobileraker/service/moonraker/file_service.dart:192)
<asynchronous suspension>
#13     FilesPageController.fetchDirectoryData (package:mobileraker/ui/screens/files/files_controller.dart:79)
<asynchronous suspension>

```

Clon1998, 2023-06-28:
Can you share a screenshot of your printer address and websocket address (If you want you can obfuscate the first 3-6 digits.

Druntar, 2023-07-23:
Fresh install seeing this same error on my phone. Hardware is a raspberry pi 3b+.

Clon1998, 2023-07-23:
I think I resolved this issue in 93f9675f8e2630311659325996a3d8f3422d6196.

However, to verify that can either of you please provide me with the output that is shown if you visit `http://<YOUR PRINTER IP/HOST>/server/files/directory?path=gcodes&extended=true`

Clon1998, 2023-08-11:
https://github.com/Clon1998/mobileraker/issues/

closed 2023-08-11 by Clon1998

Danix6, 2023-11-25:
Hello
I have the same problem on Android phone
Here is error dump
`Exception:
 type 'String' is not a subtype of type 'num' in type cast

#0      _$$GCodeThumbnailImplFromJson (package:common/data/dto/files/gcode_thumbnail.g.dart:11)
#1      new _$GCodeThumbnailImpl.fromJson (package:common/data/dto/files/gcode_thumbnail.freezed.dart:154)
#2      _$GCodeThumbnailFromJson (package:common/data/dto/files/gcode_thumbnail.freezed.dart:18)
#3      new GCodeThumbnail.fromJson (package:common/data/dto/files/gcode_thumbnail.dart:30)
#4      _sortedThumbnails.<anonymous closure> (package:common/data/dto/files/gcode_file.dart:131)
#5      MappedIterable.elementAt (dart:_internal/iterable.dart:375)
#6      ListIterator.moveNext (dart:_internal/iterable.dart:344)
#7      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189)
#8      new _GrowableList.of (dart:core-patch/growable_array.dart:150)
#9      new List.of (dart:core-patch/array_patch.dart:47)
#10     IterableExtension.sortedBy (package:collection/src/iterable_extensions.dart:66)
#11     _sortedThumbnails (package:common/data/dto/files/gcode_file.dart:132)
#12     _$$GCodeFileImplFromJson (package:common/data/dto/files/gcode_file.g.dart:40)
#13     new _$GCodeFileImpl.fromJson (package:common/data/dto/files/gcode_file.freezed.dart:439)
#14     _$GCodeFileFromJson (package:common/data/dto/files/gcode_file.freezed.dart:18)
#15     new GCodeFile.fromJson (package:common/data/dto/files/gcode_file.dart:103)
#16     FileService._parseDirectory.<anonymous closure> (package:common/service/moonraker/file_service.dart:373)
#17     new _GrowableList.generate (dart:core-patch/growable_array.dart:136)
#18     FileService._parseDirectory (package:common/service/moonraker/file_service.dart:369)
#19     FileService.fetchDirectoryInfo (package:common/service/moonraker/file_service.dart:232)
<asynchronous suspension>
#20     FutureHandlerProviderElementMixin.handleFuture.<anonymous closure>.<anonymous closure> (package:riverpod/src/async_notifier/base.dart:348)
<asynchronous suspension>`

vcdx71, 2023-12-01:
> I think I resolved this issue in [93f9675](https://github.com/Clon1998/mobileraker/commit/93f9675f8e2630311659325996a3d8f3422d6196).
> 
> However, to verify that can either of you please provide me with the output that is shown if you visit `http://<YOUR PRINTER IP/HOST>/server/files/directory?path=gcodes&extended=true`

Seeing this issue as well. Going to that URL produces a valid json (but obviously only if I add the correct port 4408 in my case)
http://192.168.2.134:4408/server/files/directory?path=gcodes&extended=true

Latest iOS app (2.6.7) and moonraker version v0.8.0-204-gfb15b2a3

Error I see in the app is:

Exception:
 type 'String' is not a subtype of type 'num' in type cast

#0      _$$GCodeThumbnailImplFromJson (package:common/data/dto/files/gcode_thumbnail.g.dart:11)
#1      new _$GCodeThumbnailImpl.fromJson (package:common/data/dto/files/gcode_thumbnail.freezed.dart:154)
#2      _$GCodeThumbnailFromJson (package:common/data/dto/files/gcode_thumbnail.freezed.dart:18)
#3      new GCodeThumbnail.fromJson (package:common/data/dto/files/gcode_thumbnail.dart:30)
#4      _sortedThumbnails.<anonymous closure> (package:common/data/dto/files/gcode_file.dart:131)
#5      MappedIterable.elementAt (dart:_internal/iterable.dart:375)
#6      ListIterator.moveNext (dart:_internal/iterable.dart:344)
#7      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189)
#8      new _GrowableList.of (dart:core-patch/growable_array.dart:150)
#9      new List.of (dart:core-patch/array_patch.dart:47)
#10     IterableExtension.sortedBy (package:collection/src/iterable_extensions.dart:66)
#11     _sortedThumbnails (package:common/data/dto/files/gcode_file.dart:132)
#12     _$$GCodeFileImplFromJson (package:common/data/dto/files/gcode_file.g.dart:40)
#13     new _$GCodeFileImpl.fromJson (package:common/data/dto/files/gcode_file.freezed.dart:439)
#14     _$GCodeFileFromJson (package:common/data/dto/files/gcode_file.freezed.dart:18)
#15     new GCodeFile.fromJson (package:common/data/dto/files/gcode_file.dart:103)
#16     FileService._parseDirectory.<anonymous closure> (package:common/service/moonraker/file_service.dart:373)
#17     new _GrowableList.generate (dart:core-patch/growable_array.dart:136)
#18     FileService._parseDirectory (package:common/service/moonraker/file_service.dart:369)
#19     FileService.fetchDirectoryInfo (package:common/service/moonraker/file_service.dart:232)
<asynchronous suspension>
#20     FutureHandlerProviderElementMixin.handleFuture.<anonymous closure>.<anonymous closure> (package:riverpod/src/async_notifier/base.dart:348)
<asynchronous suspension>

Clon1998, 2023-12-01:
@vcdx71 
The issue you are describing fits #288.
Do you mind sharing the json of the endpoint `http://192.168.2.134:4408/server/files/directory?path=gcodes&extended=true` with me in the thread of #288?