Clon1998/mobileraker #192: Only one camera configuration is synced from Mainsail, even if there are multiple printers configured

gwbrown, 2023-07-01:
**Describe the bug**
I have two printers, an SV06 (Haphaestus) and and SV06 Plus (Vulcan), each with their own webcam. Mainsail is configured so that each printer is set up with its own camera. But when I look at the printers in Mobileraker, only the camera for Haphaestus is available, and is used for both printers, despite Mainsail showing each printer with its own camera, and not allowing one printer's camera to be configured as a camera for the other printer. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set up two printers on one Klipper host machine.
2. Set up a webcam via Crowsnest for each of the above printers.
3. Configure mainsail to use the appropriate camera for each printer.
4. Add the two printers to Mobileraker.
5. Observe that the same camera feed is displayed by Mobileraker for both printers, both on the overview page and the detail pages for each printer, despite each having their own camera.
6. Open the settings for each printer and observe that only the camera configured for one printer is available, and the same camera is available for both printers.

**Expected behavior**
The printers should be displayed with their own camera feed.

**Screenshots**
I recorded [a short video demonstrating the difference in behavior](https://www.youtube.com/watch?v=9kpGB0N1WNg) between Mobileraker and Mainsail.

**Desktop (please complete the following information):**
 - OS: iOS/Mac
 - Browser Firefox, iOS Safari
 - Version 114.0.2, bundled

**Smartphone (please complete the following information):**
 - Device: iPhone 11 Pro
 - OS: [e.g. iOS8.1]
 - Browser: Safari
 - Version bundled

**Mobileraker-Version
2.4.2

**Additional context**
In case it's relevant, here's my crowsnest config:
```
[crowsnest]
log_path: /home/athena/haphaestus_data/logs/crowsnest.log
log_level: debug                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam haphaestus-cam]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/v4l/by-id/usb-046d_HD_Webcam_C615_B3E5B950-video-index0
resolution: 864x480
max_fps: 10                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:
v4l2ctl: focus_automatic_continuous=0,focus_absolute=83

[cam vulcan-cam]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8555                         # Set different ports for each device!
port: 8081                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_0AF3936F-video-index0
resolution: 864x480
max_fps: 10                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:
```

Clon1998, 2023-07-01:
Hey,
can you share a screenshot of the webcam setting for each Printer in mobileraker?

gwbrown, 2023-07-01:
Certainly. Here are screenshots of the Mobileraker config pane for both Haphaestus and Vulcan.
[image]
[image]

For comparison, here are screenshots of the same config in Mainsail:
[image]
[image]
[image]
[image]

And here are screenshots of Mobileraker editing the config for Vulcan to add a webcam, and it showing up under Haphaestus in Mainsail:
[image]
[image]

gwbrown, 2023-07-01:
For what it's worth, I added these printers via OctoEverywhere integration.

torhne, 2023-08-25:
I believe you are missing a / after webcam2 - 

It should be /webcam2**/**?action=stream   
Yours is /webcam2?action=stream

Clon1998, 2023-08-25:
Thanks, @torhne for highlighting that.
I also had a look at the issue but so far I have been unable to reproduce it.
I suspect an issue with your configuration in general rather than an issue with the app.

Rick88Stang, 2023-09-10:
I'm having a similar issue. Both of my printers are running off on one Pi with 2 different webcams. I added both the printers to Mobileraker via Octoeverywhere and the cams work fine in Octoeverywhere. Through the Mobileraker iPhone app, I can only see one stream, I even manually added both cameras and on one of the printers I can switch between the cams but whatever cam is set for one shows on both. I tried adding the second cam to Mobileraker on the other printer and it will not save for some reason, It keeps defaulting back to my webcam2 no matter if I put in webcam?action. webcam/?action, or webcam2/?action

Clon1998, 2023-09-10:
> I'm having a similar issue. Both of my printers are running off on one Pi with 2 different webcams. I added both the printers to Mobileraker via Octoeverywhere and the cams work fine in [...]

I am not sure if I understood everything correctly.
Let me summarize the things I understood.
1. You are running two printers from a single Raspberry Pi. 
2. Each printer has its own Camera, connected to the same Pi (Q: What are the URLs of these cams)
3. You added both printers to Mobileraker directly via the Octoeverywhere Buttons. Mobileraker is able to connect both printers just fine.

Not to the things I did not get correctly:
What exactly is wrong with the webcams? Could you write it down in a little bit more structured way?

E.g.:

**Printer A:**
- Cam-One is visible and working 
- Also shows Cam-Two that is supposed to be for printer B
...

**Printer B**:
- No cam at all
...

Rick88Stang, 2023-09-10:
1,2, and 3 are correct. I tried adding the URLs of the cams but I'll reference them here, this is also how they are setup on the raspberry pi and accessed via Fluidd.

Cam 1: 
- 192.168.1.13:7125/webcam/?action=stream (also just tried /webcam/?action=stream and /webcam?action=stream)
- Assigned to Printer A

Cam 2: 
- 192.168.1.13:7126/webcam2/?action=stream (also just tried /webcam2/?action=stream)
- Assigned to Printer B

Printer A:
- Cam-2 is visible and working (showing the wrong cam)
- No matter how I try to add Cam-1 it will not add and reverts to Cam-2

Printer B:
- Cam-1 &  Cam-2 are visible and working using the dropdown menu

Whatever Cam I assign to Printer B only the feed from Cam-2 is visible on the overview page for both tiles.

I hope that clears some stuff up. I'm sorry, my wife has covid so I'm a bit scatterbrained right now.

Clon1998, 2023-09-10:
> 1,2, and 3 are correct. I tried adding the URLs of the cams but I'll reference them here, this is also how they are setup on the raspberry pi and accessed via Fluidd.
>
>
>
> Cam 1:
>
> - [...]

No worries at all! And thanks for the explanation.

Could you upload the mobileraker logs after you tried to add the missing/correct cam to printer A.
The logs are located at the bottom of the app settings page within mobileraker.

Clon1998, 2024-01-19:
Closed due to inactivity for two months. Feel free to reopen if additional information is provided.

closed 2024-01-19 by Clon1998