Clon1998/mobileraker #193: Allows custom CAs

Clon1998, 2023-07-01:
Rather than just enabling the handling of self-signed certificates by overwriting badCertification, it would be better to give users the option of uploading 1..n custom CAs to accommodate a wider range of scenarios. However, it is also important to retain the ability to overwrite badCertification for maximum flexibility. It is important to note that only one of these options should be enabled at a time.

This will require the usage of a `SecurityContext` when creating either an HTTP or WS client

TheTipo01, 2023-07-01:
Wouldn't a simple popup letting the user trust the certificate presented by the web server suffice? Obviously while showing all the cert info, including the fingerprint.

Clon1998, 2023-07-02:
> Wouldn't a simple popup letting the user trust the certificate presented by the web server suffice? Obviously while showing all the cert info, including the fingerprint.

I will see how I do it. 
Probably I will offer both options to tue User. However, the percentage of users to whom these feature is useful is very low anyway.

commit 8377c223 (2023-12-07): Closes #193, #280 - Allowed user to define ssl-cert for pinning