Clon1998/mobileraker #194: Notification icon doesn't display correctly/at all

bengineer0013, 2023-07-06:
**Describe the bug**
The square app icon in the notification doesn't expand to fill the circular space. When the notification is then expanded, the icon is replaced by a white square. 
[see screenshots]

**To Reproduce**
1. Simply have the printer send out a notification.

**Expected behavior**
The square logo should fill out the circle. When the notification is expanded, the icon should not change/disappear.

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Xiaomi Redmi Note 10 Pro
 - OS: MIUI 14.0.3 Android 13

**Mobileraker-Version**
2.4.2-219

Clon1998, 2023-07-07:
Hey thanks for the bug report. I will look into that. I think I just referenced the wrong image here.

Clon1998, 2023-08-11:
Should be fixed in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0).
Please let me know if it works as soon as the new version is available trough google play (Might take 24-48 hrs)

closed 2024-03-21 by Clon1998