Clon1998/mobileraker #197: Ability to change ETA from 24hour to 12hour clock

fuzelet, 2023-07-13:
**Is your feature request related to a problem? Please describe.**
I live in a country that uses 12 hour times such as 4:00PM. While it is not hard to change 24 hour clocks to AM/PM times, it would be nice if there was a feature to change the display shown in the ETA section of the application.

**Describe the solution you'd like**
Be able to show the ETA in 12 hour clock format instead of 24 hour clock.

**Describe alternatives you've considered**
Using a browser to check status of printer using the Mainsail webpage.

**Additional context**
ETA in Mainsail:
[image]

ETA in Mobileraker:
[image]

Clon1998, 2023-07-13:
Sure, adding that should be straightforward. I'll double-check to make sure that all time displays use either the 24 or 12-hour format, depending on what the user prefers.

Clon1998, 2023-08-11:
Added in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0)

closed 2023-08-11 by Clon1998