Clon1998/mobileraker #198: Must change camera streaming modes multiple times to take effect

mbrady4597, 2023-07-14:
**Describe the bug**
A clear and concise description of what the bug is.
The webcam mode will not change if a new one is selected then saved. It must be selected two times from the drop down in the machine settings before saving in order for the change to occur. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings for the machine in use.
2. Click on the camera and then the camera mode drop down. Select an option different from the current one on use (I was trying to switch from webrtc to adaptive mjpeg for remote viewing) then hit save.
3. The camera mode does not update.
4. If instead I select the drop down and camera mode twice then save, the camera mode will update.

**Smartphone (please complete the following information):**
 - Device: iPhone 13 Pro 
 - OS: iOS 16.5.1

**Mobileraker-Version
2.4.2-219

Clon1998, 2023-07-14:
Do you mind sharing a screen recording of you changing the modes?
I wonder if it's just network latency that messes up the app.

mbrady4597, 2023-07-14:
https://github.com/Clon1998/mobileraker/assets/128195716/27274e31-801d-44f9-b96b-007f207364bf

psi-4ward, 2023-07-18:
Delete / create a cam is a workaround.

Clon1998, 2023-07-18:
Related to https://github.com/flutter-form-builder-ecosystem/flutter_form_builder/issues/1252

Clon1998, 2023-08-11:
Fixed in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0)

closed 2023-08-11 by Clon1998