Clon1998/mobileraker #200: Printer Refresh Error

austinrdennis, 2023-07-24:
**Describe the bug**
Occasionally moonraker will become out of sync on the client side and display stale data along with this cryptic error message in the screenshots section.

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot host Pi and it should happen. Other than that, it's totally random.

**Expected behavior**
Printer refreshes client with no errors and data is up to date.

**Screenshots**

[image]

**Smartphone (please complete the following information):**
 - Device: iPhone 14 Pro
 - OS: iOS
 - Version: 16.5.1 (c)

**Mobileraker-Version
2.4.2

**Additional context**
Not sure if this is a problem with my moonraker install in general or mobileraker, but the app is the only place that throws an error message. Mainsail is also frequently out of sync until I refresh the page if I leave if for a bit.

Clon1998, 2023-07-24:
Hey thanks for your bug report!
I have seen that error myself a few times. In my case it mostly occurs if I open the app after it was backgrounded for some time.

Because of that it is really hard for me to reproduce and diagnose the exact issue.

Let me know if that is different for you.

austinrdennis, 2023-07-24:
> 

You're most welcome, let me know if there's anything further I can do to help diagnose and troubleshoot! My software skills are quite limited, but I can get iOS development apps installed if that helps. 

That sounds exactly like what I'm experiencing too. It's not like it's a problem, I just didn't see a fix for it anywhere so I was taking a long shot here. Good to have it documented in any case.

Clon1998, 2023-08-11:
I will leave that open but as far as I can tell version [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0) fixes that issue.

austinrdennis, 2023-09-21:
I haven't seen this is quite a while. It's probably safe to close this.

closed 2023-09-21 by Clon1998