Clon1998/mobileraker #203: Mobileraker exception unable to list webcams

maydayVFR, 2023-07-28:
**Describe the bug**
Apologies if duplicate snd could very well be my own doing but underneath the webcam section on dashboard there is no webcam view and instead it is replaced with 

[image]

Delete and adding webcam has had no effect. Delete and reinstall mobileraker has had no effect. Wipe data on mobileraker and reinstall unsuccessful as well. Reset crowsnest and mobileraker installation via kiauh but same result.  Webcam is alive and well in mainsail,obico etc just bit mobileraker. 

**To Reproduce**
Steps to reproduce the behavior:
1. Always present. 

**Expected behavior**
Webcam to be displayed. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: iOS 
 - Browser:iOS app
 - Version: 2.4.3 

**Smartphone (please complete the following information):**
 - Device: iPhone 12 pro max
 - OS: 14.3
 - Browser: chrome/safari
 - Version 113.0.

**Mobileraker-Version
2.4.3

**Additional context**
Add any other context about the problem here.

austinrdennis, 2023-07-28:
I'm having the exact same issue on iOS as well, not just you. 

[image]

[image]

Clon1998, 2023-07-28:
Thanks for the report,
can you guys please provide me the json of the following GET request: `http://<PRINTER IP>/server/database/item?namespace=webcams`

bbraeu, 2023-07-28:
I have the same problem with the latest version.

Output of /server/database/item?namespace=webcams
{"result": {"namespace": "webcams", "key": null, "value": {"21ee1ae8-1545-45d2-b86b-eb28e4359699": {"uuid": "21ee1ae8-1545-45d2-b86b-eb28e4359699", "name": "Logitech", "service": "mjpegstreamer-adaptive", "flipX": 0, "flipY": 0, "urlStream": "/webcam?action=stream", "urlSnapshot": "/webcam?action=snapshot", "rotation": 0, "enabled": 1, "targetFps": 15, "targetFpsIdle": 5, "aspectRatio": "4:3", "icon": "mdiWebcam", "location": "printer", "extra_data": {}}}}}

Clon1998, 2023-07-28:
> I have the same problem with the latest version.
>
>
>
> Output of /server/database/item?namespace=webcams
>
> {"result": {"namespace": "webcams", "key": null, "value": [...]

Thanks for the response. 
What version of moonraker are you using?

bbraeu, 2023-07-28:
Mobileraker v0.4.0-21-g7bb9f8f8
Moonraker v0.8.0-125-g937c766d

Clon1998, 2023-07-28:
Alright, 
what web-UI are you using?

Clon1998, 2023-07-28:
Based on a test with Fluidd and Mainsail, it seems like Mainsail breaks the Moonraker webcam API. As a solution, kindly consider downgrading to an earlier version of Mainsail, pre-2.6.x. Alternatively, remove the webcams and re-add them using Fluidd/Mobileraker. Afterward,  avoid making any changes to the webcams through Mainsail until the problem is fixed.

I have raised a bug report in both repos of [Mainsail](https://github.com/mainsail-crew/mainsail/issues/1493) and [Moonraker](https://github.com/Arksine/moonraker/issues/694).

Clon1998, 2023-07-28:
Thanks to @Arksine this issue should be resolved.
Please update moonraker. Afterward, you guys are able to edit the cams again via Mainsail without any problem!

closed 2023-07-28 by Clon1998

maydayVFR, 2023-07-29:
Quick work as always guys. thanks!

DrFate09, 2023-07-30:
@Clon1998 I'm still having this error although I have updated to the latest Moonraker. Any help?

maydayVFR, 2023-07-30:
@DrFate09 I continued to have the issue even after updating as well. I looked everything over. Check the webcam page on your mainsail gui. Webcam url stream for me was `/webcam/webrtc` I think...while I was still getting the error. Changing to `/webcam/?action=stream` And restart crowsnest fixed it. Could be unrelated but I believe this is elftovers from the v3-v4 crowsnest upgrade issue. Hope that helps.

austinrdennis, 2023-08-01:
> @DrFate09 I continued to have the issue even after updating as well. I looked everything over. Check the webcam page on your mainsail gui. Webcam url stream for me was `/webcam/webrtc` I think...while I was still getting the error. Changing to `/webcam/?action=stream` And restart crowsnest fixed it. Could be unrelated but I believe this is elftovers from the v3-v4 crowsnest upgrade issue. Hope that helps. 
> 
> 
> 
> Clarity edit for future chrononauts : webcam  url snapshot should remain  `/webcam/?action=snapshot`

That fixed my issues as well and I'm working with a fairly fresh install of Klipper (entire host image). The golden goose would be support for the webRTC format. MJPEG just chugs for me compared to that, but it works for now so I'll take it.

Clon1998, 2023-08-01:
> That fixed my issues as well and I'm working with a fairly fresh install of Klipper (entire host image). The golden goose would be support for the webRTC format. MJPEG just chugs for me compared to [...]

That will be included in the next release. It's already available for beta testing on Android and closed testing on iOS.

maydayVFR, 2023-08-07:
> That fixed my issues as well and I'm working with a fairly fresh install of Klipper (entire host image). The golden goose would be support for the webRTC format. MJPEG just chugs for me compared to that, but it works for now so I'll take it.

it does for me as well. Running with crowsnest set to adaptive gets a high FPS boost. Also adjusting the resolution in Crowsnest.conf then I restart crowsnest for good measure, you should see some difference. In the crowsnest.conf file with `mode: camera-streamer` And `enable_rtsp: true` Rtsp is active at 
`http://192.168.(whatever your machine ip is)/webcam/?action=stream`
In what capacity on network I’m not sure I haven’t really played with it I just know it’s accessible...to me at least.

Nastras, 2023-09-16:
I get this error for my cam. Is it also related to this issue?
[image]