Clon1998/mobileraker #204: Unable to add webcam,MobilerakerException error

reboot-8, 2023-07-29:
**Describe the bug**
When trying to add a webcam I am getting an MobilerakerException error(see screenshot for full error)and not able enter any details of my rpi camera.

[image]

My camera is working fine and I can view it from mainsail or remotely via OctoEverywhere. Using Crowsnest with the rpi V2 camera.

All my conf files seem good and I've fresh installed MobileRaker with the same error.

App version is 2.4.3
Pixel 6 Android 13
Klipper setup fully updated

reboot-8, 2023-07-29:
Snip from the debug logs

```
[[38;5;12m💡 List Webcams request...[0m]
[[38;5;12m💡 Getting null[0m]
[[38;5;12m💡 <<<Received Server.Info[0m]
[[38;5;12m💡 Refreshing printer for uuid: 1d5ab0e1-a4c0-40f2-b243-cbc835c17f68[0m]
[[38;5;12m💡 >>>Querying printers object list[0m]
[[38;5;12m💡 Getting settings[0m]
[[38;5;196mtype 'int' is not a subtype of type 'bool?' in type cast[0m, [38;5;196m#0   _$WebcamInfoFromJson (package:mobileraker/data/model/moonraker_db/webcam_info.g.dart:19)[0m, [38;5;196m#1   new WebcamInfo.fromJson (package:mobileraker/data/model/moonraker_db/webcam_info.dart:57)[0m, [38;5;196m#2   WebcamInfoRepositoryImpl.fetchAll.<anonymous closure> (package:mobileraker/data/repository/webcam_info_repository_impl.dart:35)[0m, [38;5;196m#3   MapBase.map (dart:collection/maps.dart:82)[0m, [38;5;196m#4   WebcamInfoRepositoryImpl.fetchAll (package:mobileraker/data/repository/webcam_info_repository_impl.dart:32)[0m, [38;5;196m#5   <asynchronous suspension>[0m, [38;5;196m#6   WebcamService.listWebcamInfos (package:mobileraker/service/moonraker/webcam_service.dart:73)[0m, [38;5;196m#7   <asynchronous suspension>[0m, [38;5;196m#8   FutureHandlerProviderElementMixin.handleFuture.<anonymous closure>.<anonymous closure> (package:riverpod/src/async_notifier/base.dart:337)[0m, [38;5;196m#9   <asynchronous suspension>[0m, [38;5;196m⛔ Error while listing cams[0m]
```

Clon1998, 2023-07-29:
Hey thanks for bringing that up, 
this issue was caused due to a bug in Moonraker. The dev of moonraker already pushed a fix yesterday resolving that issue.
If you want to read up read this related issue https://github.com/Clon1998/mobileraker/issues/203.

closed 2023-07-29 by Clon1998