Clon1998/mobileraker #205: Exception:  type 'Null' is not a subtype of type 'String'

Turge08, 2023-07-29:
**Describe the bug**
I get this error when accessing one of my printers. Other one is fine. I have tried removing the printer and re-adding it and it appears to be an error with my config but I have compared it with the config on my other printer and couldn't find anything that stands out.
[Exception.txt](https://github.com/Clon1998/mobileraker/files/12207800/Exception.txt)

I still get notifications for the printer but can't load up the Dashboard for it.

I have taken a look at config_stepper.dart and the only non null string fields are stepper name, stepPin and dirPin and these are all defined.

**Smartphone (please complete the following information):**
 - Device: Tried 2 different phones: Samsung Galaxy S23 Ultra, S21 Ultra
 - OS: Android

**Mobileraker-Version
Version 2.4.2

**Additional context**
[image]
[image]

To simplify things, here's my config file: https://codebeautify.org/jsonviewer/y23739d84

Clon1998, 2023-07-29:
Thanks for the report and also including the error log and config already!
I will look into that and get back to you.

Clon1998, 2023-07-29:
Could you temporary try to exclude the `stepper_fan drivers_fan` config, seems like I have an error in my parsing mobileraker tries to parse the `stepper_fan` as stepper.

Turge08, 2023-07-29:
Oops. I created that component (based on controller_fan) in Klipper so that my fan would only come on based on the steppers without needing to define a heater. I created it about 2 months ago and haven't used the printer until recently which is why I didn't link the 2.

I'll remove it. Sorry about that!

Turge08, 2023-07-29:
It's working now without it :)

Clon1998, 2023-07-29:
Thats Great!
However, the app still has a problem. It should be fine to have a fan configured as steper_fan :)

Turge08, 2023-07-29:
True. As a developer, I probably would've implemented it the same way but I guess the best way to future-proof it would be to use a regex.

Clon1998, 2023-08-11:
Fixed in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0)

closed 2023-08-11 by Clon1998

busheezy, 2023-08-30:
This is still happening to me with 2.5.3-249.

I've added a heater_generic and now I can no longer use the app to control my printer.

Exception:
```
 type 'Null' is not a subtype of type 'String' in type cast

#0      _$$_ConfigHeaterGenericFromJson (package:mobileraker/data/dto/config/config_heater_generic.g.dart:14)
#1      new _$_ConfigHeaterGeneric.fromJson (package:mobileraker/data/dto/config/config_heater_generic.freezed.dart:196)
#2      _$ConfigHeaterGenericFromJson (package:mobileraker/data/dto/config/config_heater_generic.freezed.dart:18)
#3      new ConfigHeaterGeneric.fromJson (package:mobileraker/data/dto/config/config_heater_generic.dart:45)
#4      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:102)
#5      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:697)
#6      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:529)
#7      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:478)
#8      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:614)
#10     PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:773)
<asynchronous suspension>
#11     PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:203)
<asynchronous suspension>
```

Failed-Key: configfile 
Raw Json:
```json
{
    "configfile": {
        "config": {
            "virtual_sdcard": {
                "path": "/home/bush/sv01_data/gcodes"
            },
            "pause_resume": {},
            "display_status": {},
            "menu __main __octoprint": {
                "type": "disabled"
            },
            "menu __main __sdcard": {
                "type": "disabled"
            },
            "menu __main __macros": {
                "type": "list",
                "name": "Macros"
            },
            "menu __main __macros __resume": {
                "type": "command",
                "enable": "{printer.print_stats.state == \"paused\"}",
                "name": "Resume printing",
                "gcode": "\nRESUME"
            },
            "menu __main __macros __pause": {
                "type": "command",
                "enable": "{printer.print_stats.state == \"printing\"}",
                "name": "Pause printing",
                "gcode": "\nPAUSE"
            },
            "menu __main __macros __cancel": {
                "type": "command",
                "enable": "{(printer.print_stats.state == \"printing\" or printer.print_stats.state == \"paused\")}",
                "name": "Cancel printing",
                "gcode": "\nCANCEL_PRINT"
            },
            "stepper_x": {
                "step_pin": "PF0",
                "dir_pin": "PF1",
                "enable_pin": "!PD7",
                "microsteps": "16",
                "rotation_distance": "40",
                "endstop_pin": "^PE5",
[... 833 lines cut ...]
        "stalls": 0,
        "estimated_print_time": 62717.8476845,
        "extruder": "extruder",
        "position": [
            0.0,
            0.0,
            0.0,
            0.0
        ],
        "max_velocity": 300.0,
        "max_accel": 3000.0,
        "max_accel_to_decel": 1500.0,
        "square_corner_velocity": 5.0
    },
    "extruder": {
        "temperature": 22.36,
        "target": 0.0,
        "power": 0.0,
        "can_extrude": false,
        "pressure_advance": 0.0,
        "smooth_time": 0.04,
        "motion_queue": null
    }
}
```