Clon1998/mobileraker #206: No Notifications again :(

Experte9, 2023-07-29:
Hi again, 

I run into Problems again:(
Over the days I got less updates and now I don't get any Updates. 
Maybe it's easy but I don't know where to start. 
What you need to help?

_Originally posted by @Experte9 in https://github.com/Clon1998/mobileraker/issues/93#issuecomment-1656928987_

Clon1998, 2023-07-29:
Can you please provide me the `mobileraker.log` file?
Afterward, please try to restart the mobileraker application via mainsail or fluid.

For Mainsail:
[image]

For Fluidd:
[image]

Experte9, 2023-07-30:
I tryed restart myany times, also reboot the pi.
here is the Log:
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/12209577/mobileraker.log)

Clon1998, 2023-07-30:
Hey,
thanks for the logs.
What kind of notifications are you expecting for your machines?

Judging by your logs, you only have the status notification of type `Completed` enabled while print-progress notifications are disabled entirely. Therefore, the companion is working as expected.

Please double check your config in the App's setting page and if that shows something different, please post a screenshot here.

Experte9, 2023-07-30:
Yeah I only have Complete as Notification. But I don't get this one. The App is open in the Background.

Clon1998, 2023-07-30:
Can you please issue a custom notification to see if the notifications themself are working:
Please run this in the GCode console:
`M117 $MR$:Test notification. Testing 123123`

Experte9, 2023-07-30:
The test Notification worked. And after this I also got other. 
[image]

Is it normal that I get start and 0% progress? 
I have set up notification for "printing", "complete" and 50% progress atm. 
[image]

Can I leave this Ticket open for 2 days? I want to check if it work over days.

Clon1998, 2023-07-30:
> The test Notification worked. And after this I also got other. 
> [image]
> 
> Is it normal that I get start and 0% progress? 
> I have set up notification for "printing", "complete" and 50% progress atm. 
> [image]
> 
> Can I leave this Ticket open for 2 days? I want to check if it work over days. 

Sure I can keep this ticket open.

It's normal the 50% steps just means the app will send a notification at 0%, 50% and 100% (But i think 100% is just the completed notification)

This is due to the progress notification bar I can hopefully use in the near future for notifications on Android and Ios.

Experte9, 2023-07-31:
So yesterday and this morning I got all notifications. 
Then I switched back to Notification when finished aaand nothing. 
Only the "progress" notification tell me that the print is finished.

Clon1998, 2023-08-01:
> So yesterday and this morning I got all notifications. 
> Then I switched back to Notification when finished aaand nothing. 
> Only the "progress" notification tell me that the print is finished. 

In that case, can you provide me the log after a print finished and you didn't get s notification?

Experte9, 2023-08-01:
Don't know how to tell, but today it worked great. 
Do I get notifications on mobile Network?

Clon1998, 2023-08-01:
> Don't know how to tell, but today it worked great. Do I get notifications on mobile Network?

Yes, you should receive notifications whether you are cellular or Wi-Fi. However, there is the possibility that the OS prevents the delivery of notifications due to the energy-saving policies of the IOS.
The next time you notice that the notification was not delivered, kindly provide me the `mobileraker.log`. I am kinda curious now about what exactly is dropping the notification. Either it is the companion, my backend, or Apple/IOS. With another log, I can further investigate.

Experte9, 2023-08-02:
So my last print finished yesterday at 22:30 (see telegram bot) and I dont get a notification. My Phone was between 30-25% battery and not in energy safe mode.
Here is the log (I think it looks good);
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/12236119/mobileraker.log)
[image]

Experte9, 2023-08-02:
I don't know if this help you but I wanted to print now and open the App and saw the last status from the print (start with 0%)
[image]

Clon1998, 2023-08-02:
Hey thanks for the log.
It seems like the companion noticed that the print state changed from printing to complete and also triggered the notification creation process. However, no notification was created. I will look into this today and either find the bug that prevents the notification creation or push a update that contains more log entries in order to debug this even more.

Experte9, 2023-08-03:
Today again no notification
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/12248951/mobileraker.log)
I open the app at 98% and leave it in the Background.
Last restart pi: yesterday

If there is anything else I can help with, let me know

Clon1998, 2023-08-03:
> Today again no notification
> 
> [mobileraker.log](https://github.com/Clon1998/mobileraker/files/12248951/mobileraker.log)
> 
> I open the app at 98% and leave it in the Background.
> 
> Last restart pi: yesterday
> 
> 
> 
> If there is anything else I can help with, let me know

Hey,
i had a look at the log from yesterday and didn't find anything in particular that could prevent the notification delivery.
I will get back to you by the end of the day with an url that accesses the id stored on your printer to deliver push notifications. Maybe there is something wrong with it.

Clon1998, 2023-08-03:
Actually I have the query available on my phone as well.

Please send me the output of this html request:

http://<Your printer Ip>/server/database/item?namespace=mobileraker

Experte9, 2023-08-03:
Okay her is the Code:
`{"result": {"namespace": "mobileraker", "key": null, "value": {"fcm": {"3abbdfb2-4615-4dc8-b01d-819aad390e00": {"created": "2023-02-19T14:34:29.644676", "lastModified": "2023-07-30T20:39:44.100636", "fcmToken": "cr-OUz4eBEjPmHhIcKyzyV:APA91bEw1uD0Tf9G8Oz2rlYQErB7q1wObX_t-D3XGgjsuA9lpuM6MBhYCuofv-2OjmI5T6yCPNlzUFHTgBZI1rjRXoJ8oGxhwNdo1MXFAlW7WDiOxQq4nX_x1OreWwlzYVAEWp-uS0r4", "machineName": "Ender_wei\u00df", "language": "en", "settings": {"created": "2023-02-19T14:34:29.644678", "lastModified": "2023-08-01T16:42:31.214589", "progress": -1.0, "states": ["complete"]}, "snap": {"progress": 0.0, "state": "complete", "m117": ""}}, "client": {"lastSeen": "2023-08-03T10:08:41.051753", "version": "v0.4.0-21-g7bb9f8f"}}, "fcmTokens": [], "printerId": "29984e72-5bdc-469c-8914-c52d5319a20f", "settings": {"created": "2022-12-09T12:54:36.766618", "lastModified": "2023-02-01T20:55:42.872067", "inverts": [false, false, false], "speedXY": 50, "speedZ": 30, "extrudeFeedrate": 5, "moveSteps": [1, 10, 25, 50], "babySteps": [0.005, 0.01, 0.05, 0.1], "extrudeSteps": [1, 10, 25, 50], "macroGroups": [{"created": "2023-02-01T20:55:42.871994", "lastModified": "2023-02-01T20:55:42.871994", "name": "Default", "uuid": "96fdb5e9-7454-4cfd-b1e9-38b91c2195d3", "macros": [{"created": "2023-01-21T12:56:00.752317", "lastModified": "2023-01-21T12:56:00.752317", "name": "POWER_OFF_PRINTER", "uuid": "4cdb1a4c-5a70-4b81-8ce7-5f186e0f6d49", "visible": true, "showWhilePrinting": true}]}], "temperaturePresets": [{"created": "2023-01-06T19:45:36.602361", "lastModified": "2023-02-01T20:55:42.872019", "name": "PLA", "uuid": "e1da048a-1063-4f1b-9767-53150434fd9a", "bedTemp": 60, "extruderTemp": 200}]}}}}`

I dont know if this important to you but I use Multiklipper (2 Printer on 1 Pi)

Clon1998, 2023-08-03:
Hey, thanks for that.
I think I finally found the bug that is causing that behavior. If a user only has the `complete` state active, the companion won't update the so-called snapshot in the database. Because of that, you only get a notification if you shut down the printer/restart it. For now please make sure to have both `complete & printing` active to reset this database entry if you start a new print. I will need to look into this when I have more time to fix it properly.

Experte9, 2023-08-04:
Thanks for the temporary fix. 
It is working now, see pictures.

[image]
[image]

closed 2023-08-11 by Clon1998