Clon1998/mobileraker #207: Public / Remote URL not working

Ltek, 2023-07-30:
**Describe the bug**
the app fails to connect to a printer via a remote (internet based) URL. example... http://MyPrinter.MyDomain.COM
I can connect directly via browser and using other apps, like Klipperoid. I prefer Mobileraker if I can use it.

**To Reproduce**
use public URL in "Printer - Address" setting

**Expected behavior**
connects to the printer

**Screenshots**
n/a

**Desktop (please complete the following information):**
n/a

**Smartphone (please complete the following information):**
  - Device: Samsung S23
  - OS: Android

**Mobileraker-Version
2.4.3

Clon1998, 2023-07-30:
Could you try please attach the log of Mobileraker.
Also what error is the app showing?

Ltek, 2023-08-07:
> Could you try please attach the log of Mobileraker. Also what error is the app showing?

I dont know where the log would be. I looked in all the menus and didnt see a way to DL the log.

I'm using the same settings and API key I do with Klipperdroid. But Klipperdroid doesnt have a websocket setting
Here's screenshot of error...

[image]

Clon1998, 2023-08-07:
Hey,
getting the logs is still a bit tricky. I will have to change that in the future. You can find them at the bottom of the *App-Settings* page -> *Deug-logs*.

However, since you are seeing a 400-Status code Id assume that you configured one of the address fields wrong.
In most cases, the fields should have these values (Unless you changed the endpoint for the WebSocket server):

**Printer-Adress:** `http://myprinter.mydomain.com/`
**Websocket-Adress:** `ws://myprinter.mydomain.com/websocket`

If you are using a custom port e.g. 7199:
**Printer-Adress:** `http://myprinter.mydomain.com:7199/`
**Websocket-Adress:** `ws://myprinter.mydomain.com:7199/websocket`

Ltek, 2023-08-07:
Thx, found the logs (below, cleansed). Looks like the issue is the Websocket. I use a Reverse Proxy for about a dozen apps, all work fine except this one. Proxy is on port 80. 

[ [38;5;12m💡 Started initializeAvailableMachines [0m]
[ [38;5;12m💡 Selecting machine X2 [0m]
[ [38;5;196m⛔ machineProvider creation STARTED d844890b-99b0-475c-818c-595109713bf3 [0m]
[ [38;5;196m⛔ machineProvider creation DONE d844890b-99b0-475c-818c-595109713bf3 [0m]
[ [38;5;12m💡 initialized all machineProviders [0m]
[ [38;5;12m💡 Init for X2(d844890b-99b0-475c-818c-595109713bf3) [0m]
[ [38;5;12m💡 No OE config was found! Will only rely on local client. ref:157892916 [0m]
[ [38;5;12m💡 JsonRpcClient (ws://DOMAIN-REMOVED/websocket , ClientType.local) CREATED!! [0m]
[ [38;5;12m💡 [750859198]Trying to connect to ws://DOMAIN-REMOVED/websocket [0m]
[ [38;5;12m💡 [750859198-ws://DOMAIN-REMOVED/websocket] ClientState.disconnected ➝ ClientState.connecting [0m]
[ [38;5;12m💡 Finished initializeAvailableMachines [0m]
[ [38;5;12m💡 JsonRpcClient (ws://DOMAIN-REMOVED/websocket, ClientType.local) disposed! [0m]
[ [38;5;12m💡 Notifications are permitted: false [0m]
[ [38;5;12m💡 Setup ReceiverPort! [0m]
[ [38;5;12m💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-02T02:59:38.000Z, originalAppUserId: $RCAnonymousID:API-REMOVED, allExpirationDates: {}, requestDate: 2023-06-02T02:59:38.000Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null) [0m]
[ [38;5;12m💡 RCat ID: $RCAnonymousID:API-ID [0m]
[ [38;5;12m💡 Checking for expired subs! [0m]
[ [38;5;12m💡 Theme selected: 0, available theme len: 4 [0m]
[ [38;5;12m💡 Token from FCM TOKEN-REMOVED[0m]
[ [38;5;12m💡 X2(ws://DOMAIN-REMOVED/websocket) Device's FCM token: TOKEN-REMOVED [0m]
[ [38;5;12m💡 No OE config was found! Will only rely on local client. ref:996505350 [0m]
[ [38;5;12m💡 JsonRpcClient (ws://DOMAIN-REMOVED/websocket , ClientType.local) CREATED!! [0m]
[ [38;5;12m💡 [9030432]Trying to connect to ws://DOMAIN-REMOVED/websocket [0m]
[ [38;5;12m💡 [9030432-ws://DOMAIN-REMOVED/websocket] ClientState.disconnected ➝ ClientState.connecting [0m]
[ [38;5;208m⚠️ Got channel error TimeoutException after 0:00:05.000000: Future not completed [0m]
[ [38;5;208m⚠️ Got channel error TimeoutException after 0:00:05.000000: Future not completed [0m]
[ [38;5;196m⛔ [ws://DOMAIN-REMOVED/websocket9030432] WS-Stream error: TimeoutException after 0:00:05.000000: Future not completed [0m]
[ [38;5;12m💡 [9030432-ws://DOMAIN-REMOVED/websocket] ClientState.connecting ➝ ClientState.error [0m]
[ [38;5;208m⚠️ [X2@ws://DOMAIN-REMOVED/websocket]Unable to propagate new notification settings because JRPC was not connected! [0m]
[ [38;5;208mMobilerakerException{Machine not connected, parentException: null, parentStack: null} [0m, [38;5;208m⚠️ Error while trying to fetch CompanionMeta for machine X2 (d844890b-99b0-475c-818c-595109713bf3) [0m]
[ [38;5;12m💡 Show Dialog request for DialogType.logging [0m]

Clon1998, 2023-08-10:
> Thx, found the logs (below, cleansed). Looks like the issue is the Websocket. I use a Reverse Proxy for about a dozen apps, all work fine except this one. Proxy is on port 80. I found a web site article showing how to configure Websocket in Synology (https://mlohr.com/websockets-for-synology-dsm/)
> 
> 
> 
> Love the app!  
> 
> 
> 
> Two small suggestions...
> 
> 1. on the page where you configure the settings, alert the user that Websockets may need to be separately configured to pass on firewalls, proxies, etc.
> 
> 2. Ideally we'd have two configurations per device, Local and Remote, and it would detect local based on the network name we put in... thus provide us the most efficient connection if we are Local.
> 
> 
> 
> 
> 
> thx for the quick replay and help solving!
> 
> 
> 
> LOG below...
> 
> ```
> 
> 
> 
> [ [38;5;12m💡 Started initializeAvailableMachines [0m]
> 
> [ [38;5;12m💡 Selecting machine X2 [0m]
> 
> [ [38;5;196m⛔ machineProvider creation STARTED d844890b-99b0-475c-818c-595109713bf3 [0m]
> 
> [ [38;5;196m⛔ machineProvider creation DONE d844890b-99b0-475c-818c-595109713bf3 [0m]
> 
> [ [38;5;12m💡 initialized all machineProviders [0m]
> 
> [ [38;5;12m💡 Init for X2(d844890b-99b0-475c-818c-595109713bf3) [0m]
> 
> [ [38;5;12m💡 No OE config was found! Will only rely on local client. ref:157892916 [0m]
> 
> [ [38;5;12m💡 JsonRpcClient (ws://DOMAIN-REMOVED/websocket , ClientType.local) CREATED!! [0m]
> 
> [ [38;5;12m💡 [750859198]Trying to connect to ws://DOMAIN-REMOVED/websocket [0m]
> 
> [ [38;5;12m💡 [750859198-ws://DOMAIN-REMOVED/websocket] ClientState.disconnected ➝ ClientState.connecting [0m]
> 
> [ [38;5;12m💡 Finished initializeAvailableMachines [0m]
> 
> [ [38;5;12m💡 JsonRpcClient (ws://DOMAIN-REMOVED/websocket, ClientType.local) disposed! [0m]
> 
> [ [38;5;12m💡 Notifications are permitted: false [0m]
> 
> [ [38;5;12m💡 Setup ReceiverPort! [0m]
> 
> [ [38;5;12m💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-02T02:59:38.000Z, originalAppUserId: $RCAnonymousID:API-REMOVED, allExpirationDates: {}, requestDate: 2023-06-02T02:59:38.000Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null) [0m]
> 
> [ [38;5;12m💡 RCat ID: $RCAnonymousID:API-ID [0m]
> 
> [ [38;5;12m💡 Checking for expired subs! [0m]
> 
> [ [38;5;12m💡 Theme selected: 0, available theme len: 4 [0m]
> 
> [ [38;5;12m💡 Token from FCM TOKEN-REMOVED[0m]
> 
> [ [38;5;12m💡 X2(ws://DOMAIN-REMOVED/websocket) Device's FCM token: TOKEN-REMOVED [0m]
> 
> [ [38;5;12m💡 No OE config was found! Will only rely on local client. ref:996505350 [0m]
> 
> [ [38;5;12m💡 JsonRpcClient (ws://DOMAIN-REMOVED/websocket , ClientType.local) CREATED!! [0m]
> 
> [ [38;5;12m💡 [9030432]Trying to connect to ws://DOMAIN-REMOVED/websocket [0m]
> 
> [ [38;5;12m💡 [9030432-ws://DOMAIN-REMOVED/websocket] ClientState.disconnected ➝ ClientState.connecting [0m]
> 
> [ [38;5;208m⚠️ Got channel error TimeoutException after 0:00:05.000000: Future not completed [0m]
> 
> [ [38;5;208m⚠️ Got channel error TimeoutException after 0:00:05.000000: Future not completed [0m]
> 
> [ [38;5;196m⛔ [ws://DOMAIN-REMOVED/websocket9030432] WS-Stream error: TimeoutException after 0:00:05.000000: Future not completed [0m]
> 
> [ [38;5;12m💡 [9030432-ws://DOMAIN-REMOVED/websocket] ClientState.connecting ➝ ClientState.error [0m]
> 
> [ [38;5;208m⚠️ [X2@ws://DOMAIN-REMOVED/websocket]Unable to propagate new notification settings because JRPC was not connected! [0m]
> 
> [ [38;5;208mMobilerakerException{Machine not connected, parentException: null, parentStack: null} [0m, [38;5;208m⚠️ Error while trying to fetch CompanionMeta for machine X2 (d844890b-99b0-475c-818c-595109713bf3) [0m]
> 
> [ [38;5;12m💡 Show Dialog request for DialogType.logging [0m]
> 
> ```

So you managed to resolve the issue 🤔?
I will think about adding an option to configure a remote and local address. Actually, I just reworked the entire add printer flow/UI that now offers an expert and beginner mode to making it easier for people to use. Especially the easy mode takes away many of the more advanced options and also auto configures the websocket address based on the url the user enters.

Ltek, 2023-08-10:
> [ [38;5;208m⚠️ ***@***.***://DOMAIN-REMOVED/websocket]Unable to propagate new notification settings because JRPC was not connected! [0m]

Yes, it's working now. Might good to add a note in the wiki letting users
know they may need to configure their proxy or firewall specifically to
allow websockets. I didn't know and I'm a good techie :)

Yeah, a separate way to configure remote vs local would be great!

closed 2023-08-11 by Clon1998