Clon1998/mobileraker #208: Bed Screw Adjust: Buttons out of frame in small screen device.

GAZ082, 2023-07-30:
I have a Sony XZ1 Compact. Has a 4.6" screen, 720p resolution.
[image]

The buttons are outside the modal and I cant press them :(

Clon1998, 2023-07-30:
Hey, 
Thank you for reporting the bug. It appears that I made a small mistake by forgetting to add shrinkage to the X/Y plane. I have already implemented a fix for this issue and it will be included in the next major release!

GAZ082, 2023-08-06:
@Clon1998 hello Patrick I don't want to be a pain in the ass but wheb are you releasing the next version because this bug is making the app less useful for me right now 

Thanks!

Clon1998, 2023-08-06:
> @Clon1998 hello Patrick I don't want to be a pain in the ass but wheb are you releasing the next version because this bug is making the app less useful for me right now
> 
> Thanks!

It still will take another week or two. Found a bug in the companion that requires my attention.
However, you can sideload a beta version that includes the hotfix if you need it right now (Also kindly confirm if the dialog is fixed)

https://github.com/Clon1998/mobileraker/releases/tag/pre-2.5.0

https://github.com/Clon1998/mobileraker/releases/download/pre-2.5.0/preview_b_2_5_0.apk

GAZ082, 2023-08-06:
For some reason is not installing. I made sure to have space. I'm using Android 9, perhaps the manifest in this preview requires something newer by mistake? @Clon1998

Clon1998, 2023-08-11:
Fixed in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0)

closed 2023-08-11 by Clon1998