Clon1998/mobileraker #210: App will not open on Samsung fold 5

cdeitrich00, 2023-08-09:
**Describe the bug**
The app will not open past the logo screen on the samsung galaxy fold 5

**To Reproduce**
Steps to reproduce the behavior:
1. Download app on fold 5
2. Open app
3. Wait
4. Noting loads

**Expected behavior**
The app should open and show dashboard 

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: samsung galaxy fold 5 
 - OS: android 5.1.1 
 - Browser: chrome
 - Version: 115.0.5790.166

**Mobileraker-Version: 2.4.2
This information can be found in Android's and IOS's App-Info.

**Additional context**
Add any other context about the problem here.

cdeitrich00, 2023-08-09:
Thank you for the quick reply! This seems to be working. I will set up my printer later on when I get home from work.

Clon1998, 2023-08-11:
Fixed in [android2.5.0](https://github.com/Clon1998/mobileraker/tree/android2.5.0)

closed 2023-08-11 by Clon1998