Clon1998/mobileraker #211: No remote access through Octoeverywhere with multiple printers

TheNothingMan, 2023-08-11:
**Describe the bug**
I have setup Mobileraker with two printers on the same Raspberry Pi. The printers are configured with IP and Port.
Wifi access works fine, but I can not access the printers remotely. I asked the developers from Octoeverywhere for help, and after sending them the logs, they told me that there should be no port number at the end of the Octoeverywhere URL.

**Smartphone (please complete the following information):**
 - Device: Xiaomi Mi9T Pro
 - OS: Android 11

**Mobileraker-Version
2.5

Clon1998, 2023-08-11:
Please try the latest version once it is available or download the release from [github](https://github.com/Clon1998/mobileraker/releases/tag/android2.5.0).
It should contain a fix for that issue :).

TheNothingMan, 2023-08-12:
OK, indeed, the latest version fixed it. Strange thing is that I checked Playstore for updates before opening the issue AND the app changelog said 2.5. So probably a restart somewhere finally cleared all issues.
Thank you!

closed 2023-08-12 by TheNothingMan