Clon1998/mobileraker #212: Webcam not working

TheNothingMan, 2023-08-11:
**Describe the bug**
Since version 2.3.0, my webcam doesn't show anymore. There is a 404 error.
I have configured multiple printers on the same Pi, using different ports. The webcams work in Mainsail, Octoeverywhere and Obico. I suspect that the fact that Mobileraker adds the IP **and port** before the actual webcam URL is the problem. I tried adding the webcam again from Mobileraker and Mainsail, but the error persists. Any idea what I could try next?

**Smartphone (please complete the following information):**
 - Device: Xiaomi Mi9T Pro
 - OS: Android 11

**Mobileraker-Version
2.5

Clon1998, 2023-08-11:
This issue should have been resolved with 2.5.0.
Please confirm that you are using the latest version as I just released it.

If you are on 2.5.0 please share a screenshot of the webcam configuration.

Clon1998, 2023-08-11:
Also if that does not work please upload the log files the app generates (AppSettings -> Scroll to The bottom -> Debug Logs -> Download)

TheNothingMan, 2023-08-12:
Working now - sorry, I was sure to be on the latest version.

closed 2023-08-12 by TheNothingMan

robotfishe, 2023-08-13:
I had the same issue, and it persisted with 2.5.1, but when I downloaded the debug log to post it here I found the problem: my camera settings had a relative url entered ("/webcam?action=stream") and for some reason, when Mobileraker added this to the machine IP (192.168.1.90), it was appending a port 0 and trying to contact http://192.168.1.90:0/webcam?action=stream. Putting the whole correct URL, "http://192.168.1.90/webcam?action=stream", in the settings fixed it. So if you found your way to this issue like I did by googling "Mobileraker webcam not working", try that!

Clon1998, 2023-08-13:
> I had the same issue, and it persisted with 2.5.1, but when I downloaded the debug log to post it here I found the problem: my camera settings had a relative url entered ("/webcam?action=stream") [...]

The problem you describe is related to #212. Simply put, while fixing webcams for Ocotoeverywhere I introduced an error for other webcams.