Clon1998/mobileraker #213: WebRtc camera being accessed on incorrect port.

IntenseWiggling, 2023-08-12:
**Describe the bug**
Have a webrtc (camera-streamer) camera configured in moonraker with a stream URL of `http://[IP]:8080/webrtc` . When accessing with Mobileraker 2.5.1 I get an error:

WebRtc URI: http://[IP]:0/webrtc
Connect refused

So seems it's attempting to access over port `0` instead of `8080` specified in the URI.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure webrtc camera  in moonraker, with port 8080 in the stream URI
2. Attempt to access the camera over mobileraker.
3. See error

**Expected behavior**
The camera is accessible over the specified URI.

**Screenshots**
[image]

**Mobileraker-Version
2.5.1

Clon1998, 2023-08-13:
Thanks for reporting.
I just fixed that in 4aeb3a908172a32af448d1b2180282df487a0d6c. I will wait until Thursday or Wednesday to push a hotfix version.

Clon1998, 2023-08-15:
2.5.2 should be rolling out slowly

IntenseWiggling, 2023-08-15:
Received the update earlier, and can confirm it's working great now. Thanks!