Clon1998/mobileraker #214: Mesh calibration does not work. Only does first point.

GAZ082, 2023-08-12:
Go to mesh calibration.
Calibrate the first point.
Hit Accept.

Boom, goes back to the mainscreen and asks to save config.

Version 2.5
Android 9

Clon1998, 2023-08-12:
Hey,
once again. I can not reproduce that behavior. The Bed Mesh is working just fine. Also which dialog are you talking about?
Are you talking about the Bed Screw Adjust rather than Bed mesh?

GAZ082, 2023-08-12:
Bed mesh. I'll finish a print i have now and then restart everything and try again. Will get a video if possible.

GAZ082, 2023-08-13:
https://github.com/Clon1998/mobileraker/assets/100194/031f9b3e-7a41-42ee-908a-71239e24c030

Here @Clon1998

Clon1998, 2023-08-14:
Ohh interesting.
I assume you are not trying to run auto bed level? Rather manual bed leveling where you adjust the offset via that dialog?

Tbh. manual bed leveling never was supported and that the dialog is shown is just a side effect of the probe/offset calibration call. I will look into this to ensure the dialog also works for manual bed leveling.

Thanks for reporting this!

GAZ082, 2023-08-14:
Yes! Its a manual mesh.

Thanks!

GAZ082, 2023-09-03:
Hi @Clon1998 , do you think you can do a release with the fix? Thanks!

Clon1998, 2023-09-03:
If you're on Android, there actually is a pre release on GitHub ready to download that should include the fix.

GAZ082, 2023-09-03:
Thank you Patrick! You are the best. It's working properly now. There is a small issue with my small screen device, if there is a lot of numbers the + button wants to leave the modal :D.

[image]

closed 2023-09-05 by Clon1998