Clon1998/mobileraker #215: Does not home Z axis.

GAZ082, 2023-08-12:
Hit the HOME button.
See how it homes X,Y but not Z.

Android 9
Version 2.5

Clon1998, 2023-08-12:
Hey,

I cannot reproduce that behavior for either of the homing buttons(see video). Please provide me with a screen recording. 
Also after homing please check the GCode Console. The Buttons should issue a `G28 ` command.

https://github.com/Clon1998/mobileraker/assets/10357775/f150c30c-a461-4aeb-b961-52a173d99296

GAZ082, 2023-08-12:
I'll do a more thorough test and get back to you.  The odd thing is, in Mainsail works ok.

Clon1998, 2023-08-13:
Hey,
I noticed that the after my latest change the UI can be a little disconnected from the printer data if the app is reopened after being in the backg for a while.
That means I am able to send homing commands but the UI itself does not update.
Was that the thing you've experienced or did the printer itself only home one Axis?

GAZ082, 2023-08-13:
Hey, the app is just sending

G28 XY

(The plain G28 was sent by Mainsail, the others, by your app).

[image]

GAZ082, 2023-08-13:
Oh, I'm a moron. I've been hitting the center of the Move Axis, just like in Mainsail. I just saw that there is a "ALL" button...

Will close this.

closed 2023-08-13 by GAZ082