Clon1998/mobileraker #216: Parsing of temperature_sensor failed: Null check operator used on a null value.

vaxxi, 2023-08-13:
**Describe the bug**
Error message at the bottom of the UI stating:

```
Refreshing Printer Failed
Parsing of temperature_sensor mcu_temp failed: Null check operator used on a null value.
```

``**To Reproduce**
This appears to be related to an EBB36 being installed, but I've seen it happen on the Raspberry Pi temperature monitoring as well (that one had a string of 0.0 values at the end of the raw JSON).

**Expected behavior**
No error appears.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Mobile
 - OS: Android 12
 - Browser Chrome Beta
 - Version 

**Mobileraker-Version
2.5.1

**Additional context**
```
Exception:
 Null check operator used on a null value

#0      PrinterService._updateTemperatureSensor (package:mobileraker/service/moonraker/printer_service.dart:653)
#1      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:527)
#2      PrinterService._onStatusUpdateHandler.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:511)
#3      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#4      PrinterService._onStatusUpdateHandler (package:mobileraker/service/moonraker/printer_service.dart:510)
#5      JsonRpcClient._onChannelMessage.<anonymous closure> (package:mobileraker/data/data_source/json_rpc_client.dart:273)
#6      Iterable.forEach (dart:core/iterable.dart:346)
#7      JsonRpcClient._onChannelMessage (package:mobileraker/data/data_source/json_rpc_client.dart:273)
#8      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#9      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
#10     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271)
#11     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123)
#12     _HandleErrorStream._handleData (dart:async/stream_pipe.dart:253)
#13     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153)
#14     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#15     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
#16     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271)
#17     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:776)
#18     _StreamController._add (dart:async/stream_controller.dart:650)
#19     _StreamController.add (dart:async/stream_controller.dart:598)
#20     new _WebSocketImpl._fromSocket.<anonymous closure> (dart:_http/websocket_impl.dart:1144)
#21     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#22     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
#23     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271)
#24     _SinkTransformerStreamSubscription._add (dart:async/stream_transformers.dart:63)
#25     _EventSinkWrapper.add (dart:async/stream_transformers.dart:13)
#26     _WebSocketProtocolTransformer._messageFrameEnd (dart:_http/websocket_impl.dart:332)
#27     _WebSocketProtocolTransformer.add (dart:_http/websocket_impl.dart:226)
#28     _SinkTransformerStreamSubscription._handleData (dart:async/stream_transformers.dart:111)
#29     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#30     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
#31     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271)
#32     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:776)
#33     _StreamController._add (dart:async/stream_controller.dart:650)
#34     _StreamController.add (dart:async/stream_controller.dart:598)
#35     _Socket._onData (dart:io-patch/socket_patch.dart:2381)
#36     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594)
#37     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339)
#38     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271)
#39     _SyncStreamControllerDispatch._sendData (dart:async/stream_controller.dart:776)
#40     _StreamController._add (dart:async/stream_controller.dart:650)
#41     _StreamController.add (dart:async/stream_controller.dart:598)
#42     new _RawSocket.<anonymous closure> (dart:io-patch/socket_patch.dart:1899)
#43     _NativeSocket.issueReadEvent.issue (dart:io-patch/socket_patch.dart:1356)
#44     _microtaskLoop (dart:async/schedule_microtask.dart:40)
#45     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49)

Failed-Key: temperature_sensor mcu_temp 
Raw Json:
{"temperature_sensor CANBOARD":{"temperature":44.1},"temperature_sensor mcu_temp":{"temperature":35.86},"temperature_sensor chamber":{"temperature":34.51},"toolhead":{"estimated_print_time":20.867304644444445},"extruder":{"temperature":45.04}}
```

Sensors are configured like this:
```
[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100

[temperature_sensor mcu_temp]
sensor_type: temperature_mcu
min_temp: 0
max_temp: 100
```

vaxxi, 2023-08-13:
I got the same error on the Raspberry Pi host temperature, looks like it's picking up the whole configuration files as a JSON ?

```
Exception:
 Null check operator used on a null value

#0      PrinterService._updateTemperatureSensor (package:mobileraker/service/moonraker/printer_service.dart:653)
#1      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:527)
#2      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:478)
#3      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#4      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:614)
#5      PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:773)
<asynchronous suspension>
#6      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:203)
<asynchronous suspension>

Failed-Key: temperature_sensor raspberry_pi 
Raw Json:
{"configfile":{"config":{"mcu":{"serial":"/dev/serial/by-id/usb-Klipper_stm32f.........
```

Clon1998, 2023-08-13:
Hey I will look into that.
It's not picking up the entire Json response, I just included it in the error message for easier debugging.

Clon1998, 2023-08-13:
I am not a 100% sure yet what exactly is happening. But it is not related to the json in the error.
It happens earlier because the app is not able to find the temperature sensors in the initially fetched printer data.
For debugging purposes could you please send me the out put of `<your printer ip>/printer/objects/list`

Clon1998, 2023-08-13:
I think I found the issue, for the two mentioned sensors please verify the config section to ensure it does not contain additional whitespace (` `) after the `temperature_sensor` key.
Actually, just yesterday I thought I might need to rework that logic because some users could have multiple space characters after the config name by accident which would throw an error inside the app.

Clon1998, 2023-08-14:
> Unfortunately, there is no white space after those lines, here's the `xxd` output:
> 
> ```
> 
> 00001c00: 7065 6564 3a20 312e 300a 0a23 2323 0a23  peed: 1.0..###.#
> 
> 00001c10: 2054 656d 7065 7261 7475 7265 730a 2323   Temperatures.##
> 
> 00001c20: 230a 0a5b 7465 6d70 6572 6174 7572 655f  #..[temperature_
> 
> 00001c30: 7365 6e73 6f72 c2a0 7261 7370 6265 7272  sensor..raspberr
> 
> 00001c40: 795f 7069 5d0a 7365 6e73 6f72 5f74 7970  y_pi].sensor_typ
> 
> 00001c50: 653a c2a0 7465 6d70 6572 6174 7572 655f  e:..temperature_
> 
> 00001c60: 686f 7374 0a6d 696e 5f74 656d 703a c2a0  host.min_temp:..
> 
> 00001c70: 3130 0a6d 6178 5f74 656d 703a c2a0 3130  10.max_temp:..10
> 
> 00001c80: 300a 0a5b 7465 6d70 6572 6174 7572 655f  0..[temperature_
> 
> 00001c90: 7365 6e73 6f72 c2a0 6d63 755f 7465 6d70  sensor..mcu_temp
> 
> 00001ca0: 5d0a 7365 6e73 6f72 5f74 7970 653a c2a0  ].sensor_type:..
> 
> 00001cb0: 7465 6d70 6572 6174 7572 655f 6d63 750a  temperature_mcu.
> 
> 00001cc0: 6d69 6e5f 7465 6d70 3ac2 a030 0a6d 6178  min_temp:..0.max
> 
> 00001cd0: 5f74 656d 703a c2a0 3130 300a 0a23 2323  _temp:..100..###
> 
> 00001ce0: 2323 2323 2323 2323 2323 2323 2323 2323  ################
> 
> 00001cf0: 2323 2323 2323 2323 2323 2323 2323 2323  ################
> 
> ```

I was actually taking about the Http Get earlier e.g. http://192.168.178.135/printer/objects/list

vaxxi, 2023-08-14:
Aah. That looks to be it:

```
"temperature_sensor\u00a0raspberry_pi", "temperature_sensor\u00a0mcu_temp"
```

vaxxi, 2023-08-14:
Indeed looks like something changed:

```
pi@mainsail:~/printer_data/config $ diff printer-20230812_223139.cfg printer.cfg
291c291
< [temperature_sensor raspberry_pi]
---
> [temperature_sensor raspberry_pi]
296c296
< [temperature_sensor mcu_temp]
---
> [temperature_sensor mcu_temp]
496c496
```

vaxxi, 2023-08-15:
OK, that did the trick. Must have been some non-printable characters which borked parsing. I'll close this one for now, as it's not a real bug.

closed 2023-08-15 by vaxxi