Clon1998/mobileraker #217: Hide webcam if connection to stream fails

PedroC1999, 2023-08-13:
**Is your feature request related to a problem? Please describe.**
I don't "always" have a webcam connected to the printer. When I don't, there is a large, unremovable warning in the dashboard webcam tile.

**Describe the solution you'd like**
A toggle in the settings that allows these errors to be suppressed and ignored, or an option to hide the webcam tile if no webcam connection is established

**Describe alternatives you've considered**
Any acceptable method of hiding this large error if no webcam is found

non-bin, 2024-02-11:
It'd also be nice to allow you to hide the webcam all together

I use an old phone as an interface on the printer so it just takes up space

Clon1998, 2024-02-11:
> It'd also be nice to allow you to hide the webcam all together
> 
> I use an old phone as an interface on the printer so it just takes up space

This is part of a bigger rework that will involve a customizable dashboard layout. 
All I can say is that I am working on it but there is no ETA as I am quite involved in finishing my degree right now.

non-bin, 2024-02-11:
Awesome! We all appreciate the work you've done 👍

commit 30abf13a (2024-05-23): Fix #217 - Webcams can not be "hidden" if failed. Also reworked error message layout
closed 2024-06-25 by Clon1998