Clon1998/mobileraker #221: Show camera stream(s) even when Printer (MCU) is offline.

HcNguyen111, 2023-08-17:
Can cameras be shown on the Dashboard even when the Printer/MCU is offline?

I have Klipper running on a Pi with cameras connected to the Pi. My printer isn't always powered on, but I'd still like to see the camera streams (which are on even when the printer itself isn't).

busheezy, 2023-08-31:
This would be great. I have my printers set to turn off a bit after a print finishes.

kohbo, 2023-12-05:
I'd love this

Clon1998, 2023-12-06:
"I am currently working on a major UI overhaul that will enable this feature. Unfortunately, I don't have a specific timeline for its completion yet. However, I can check the current webcam card to see if it can easily integrated into the klipper-not-ready screen.

Clon1998, 2025-01-07:
Sorry for the delay, I simply forgot about this issue.
Luckily, the UI overhaul made it insanely easy to add this feature now :)

[image]

commit 0f7e0a12 (2025-01-07): feat: Show webcam even if klipper is not connected

fix #221
closed 2025-02-17 by Clon1998