Clon1998/mobileraker #222: Cannot show/add webcam

togeis, 2023-08-17:
**Describe the bug**
Webcam not shown on Dashbaord and not possible to add the webcam. Not getting the list of available cams

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A list of webcams configured in mainsail

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: iOS

**Smartphone (please complete the following information):**
 - Device: iPhone 13
 - OS: 16.6
 

**Mobileraker-Version
2.5.1-241

Clon1998, 2023-08-18:
Hey,
thanks for reporting. Could you please provide me with your moonraker version and also the mobileraker app logs located that can be found insude the App's Settings -> Scroll to bottom -> Debug logs.

Bandit3112011, 2023-08-19:
The same error on ver. 2.5.2-246

Clon1998, 2023-08-19:
> The same error on ver. 2.5.2-246
> 
> 

Without the debug logs I can not do much. Do you mind providing them to me?

Bandit3112011, 2023-08-19:
💡 Completed FirebaseRemote init
💡 Machine in box is Sonic Pad (7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9)#379419737
💡 Completed Hive init
💡 Completed PaymentService init
💡 Selecting machine Sonic Pad
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-16T20:16:56Z, originalAppUserId: $RCAnonymousID:3e7ce37345504935932c6cafd3a78bb8, allExpirationDates: {}, requestDate: 2023-08-15T13:07:17Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:3e7ce37345504935932c6cafd3a78bb8
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Received all machines
💡 machineProvider creation STARTED 7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9
💡 machineProvider creation DONE 7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 Theme selected: 0, available theme len: 4
💡 No OE config was found! Will only rely on local client. ref:678156647
💡 JsonRpcClient (ws://192.168.178.151/websocket , ClientType.local) CREATED!!
💡 [928737998]Trying to connect to ws://192.168.178.151/websocket
💡 [928737998-ws://192.168.178.151/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
💡 Printer Service received klippyState: null
💡 Token from FCM fnpBOksz2kKfuSvtiK7I-n:APA91bF4BVnGZ3QktVdt2mxetn88nxhDw8Wuq2XSR0C_lHPjkrHr1Uqu8-pOuK_7IdZLiPTsGx_n5ckg8mFFe6ajg8K-2Yx_hzDYkHm2OV0LNWXw5HLvTjLh8DDj7Xn7WeFS3UEwjDt7
💡 Sonic Pad(ws://192.168.178.151/websocket)  Device's FCM token: fnpBOksz2kKfuSvtiK7I-n:APA91bF4BVnGZ3QktVdt2mxetn88nxhDw8Wuq2XSR0C_lHPjkrHr1Uqu8-pOuK_7IdZLiPTsGx_n5ckg8mFFe6ajg8K-2Yx_hzDYkHm2OV0LNWXw5HLvTjLh8DDj7Xn7WeFS3UEwjDt7
⚠️ Got channel error SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.178.151, port: 80
⛔ [ws://192.168.178.151/websocket928737998] WS-Stream error: SocketException: HTTP connection timed out after 0:00:03.000000, host: 192.168.178.151, port: 80
💡 [928737998-ws://192.168.178.151/websocket] ClientState.connecting ➝ ClientState.error
💡 Printer Service received klippyState: KlipperState.error
⚠️ [Sonic Pad@ws://192.168.178.151/websocket]Unable to propagate new notification settings because JRPC was not connected!
MobilerakerException{Machine not connected, parentException: null, parentStack: null}
⚠️ Error while trying to fetch CompanionMeta for machine Sonic Pad (7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9)
💡 Show Dialog request for DialogType.logging

togeis, 2023-08-19:
My moonraker version is v0.8.0-55
[mobileraker_2023-08-18T00-18-07.log](https://github.com/Clon1998/mobileraker/files/12385449/mobileraker_2023-08-18T00-18-07.log)
Logs attached

Clon1998, 2023-08-19:
@togeis thanks for that!
Your moonraker version is a bit outdated; please update it and try again in the app. If that is still not working please delete the webcams and recreate them. There was a bug in Moonraker that was addressed recently.

@Bandit3112011 your logs sadly do not contain a lot of info. Please force close the app, open it again, open the printer settings scroll down to the webcams and than send me the logs once again.

Bandit3112011, 2023-08-19:
💡 Completed FirebaseRemote init
💡 Machine in box is Sonic Pad (7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9)#413441261
💡 Completed Hive init
💡 Completed PaymentService init
💡 Selecting machine Sonic Pad
💡 Received fetchAll
💡 Got customerInfo: CustomerInfo(entitlements: EntitlementInfos(all: {}, active: {}), allPurchaseDates: {}, activeSubscriptions: [], allPurchasedProductIdentifiers: [], nonSubscriptionTransactions: [], firstSeen: 2023-06-16T20:16:56Z, originalAppUserId: $RCAnonymousID:3e7ce37345504935932c6cafd3a78bb8, allExpirationDates: {}, requestDate: 2023-08-19T09:05:35Z, latestExpirationDate: null, originalPurchaseDate: null, originalApplicationVersion: null, managementURL: null)
💡 RCat ID: $RCAnonymousID:3e7ce37345504935932c6cafd3a78bb8
💡 Received isSupporter false
💡 Max allowed machines for non Supporters is -1
💡 Completed initialRoute init
💡 Started initializeAvailableMachines
💡 Received all machines
💡 machineProvider creation STARTED 7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9
💡 machineProvider creation DONE 7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9
💡 initialized all machineProviders
💡 Completed initializeAvailableMachines
💡 Notifications are permitted: true
💡 Setup notification channels
💡 Setup ReceiverPort!
💡 Completed NotificationService init
💡 Theme selected: 0, available theme len: 4
💡 No OE config was found! Will only rely on local client. ref:1045368977
💡 JsonRpcClient (ws://192.168.178.151/websocket , ClientType.local) CREATED!!
💡 [7952861]Trying to connect to ws://192.168.178.151/websocket
💡 [7952861-ws://192.168.178.151/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
💡 Printer Service received klippyState: null
💡 [7952861-ws://192.168.178.151/websocket] ClientState.connecting ➝ ClientState.connected
💡 >>>Fetching Server.Info
💡 >>>Fetching Printer.Info
💡 <<<Received Server.Info
💡 Printer Service received klippyState: KlipperState.ready
💡 Refreshing printer for uuid: 7f5f9e69-d83f-4f57-89ff-4ae99d6e64e9
💡 >>>Querying printers object list
💡 Getting settings
💡 <<<Received Printer.Info
💡 Partial Update STARTED KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, authorization, history, octoprint_compat, timelapse], warnings: [], klippyStateMessage: null)
💡 Partial Update Done KlipperInstance(klippyConnected: true, klippyState: KlipperState.ready, components: [klippy_connection, application, websockets, internal_transport, dbus_manager, database, file_manager, klippy_apis, secrets, template, shell_command, machine, data_store, proc_stats, job_state, job_queue, http_client, announcements, webcam, extensions, authorization, history, octoprint_compat, timelapse], warnings: [], klippyStateMessage: Printer is ready)
⚠️ watchWhere just forces owner to invalidate! Ref:AutoDisposeFutureProviderElement<MachineSettings>(provider: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#a90c1, origin: selectedMachineSettingsProvider:AutoDisposeFutureProvider<MachineSettings>#a90c1)
💡 Getting settings
💡 <<<Received printer objects list!
💡 >>>Querying Printer Objects!
💡 <<<Received queried printer objects
💡 Fetching cached temperature store data
💡 Received cached temperature store for [heater_bed, extruder]
💡 Subscribing printer objects for ws-updates!
💡 Updating Default Macros for "Sonic Pad(ws://192.168.178.151/websocket)"!
💡 Getting settings
💡 Also requesting an update for current_file: 
💡 UPDATED current_file: null
💡 Max allowed machines for non Supporters is -1
💡 Last dismiss of Supporter AD: 2023-07-28 15:07:59.069
💡 List Webcams request...
💡 Trying to fetch all webcams from moonraker.
💡 Received 1 webcams from moonraker.
type 'String' is not a subtype of type 'num' in type cast
#0   _$$_WebcamInfoFromJson (package:mobileraker/data/model/moonraker_db/webcam_info.g.dart:26)
#1   new _$_WebcamInfo.fromJson (package:mobileraker/data/model/moonraker_db/webcam_info.freezed.dart:336)
#2   _$WebcamInfoFromJson (package:mobileraker/data/model/moonraker_db/webcam_info.freezed.dart:18)
#3   new WebcamInfo.fromJson (package:mobileraker/data/model/moonraker_db/webcam_info.dart:116)
#4   WebcamInfoRepositoryImpl.fetchAll.<anonymous closure> (package:mobileraker/data/repository/webcam_info_repository_impl.dart:38)
#5   MappedIterable.elementAt (dart:_internal/iterable.dart:375)
#6   ListIterator.moveNext (dart:_internal/iterable.dart:344)
#7   new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189)
#8   new _GrowableList.of (dart:core-patch/growable_array.dart:150)
#9   new List.of (dart:core-patch/array_patch.dart:47)
#10   SetBase.toList (dart:collection/set.dart:119)
#11   WebcamInfoRepositoryImpl.fetchAll (package:mobileraker/data/repository/webcam_info_repository_impl.dart:39)
#12   <asynchronous suspension>
[... 16 lines cut ...]
#6   MachineService.updateMachineFcmConfig (package:mobileraker/service/machine_service.dart:262)
#7   <asynchronous suspension>
#8   NotificationService._setupFCMOnPrinterOnceConnected (package:mobileraker/service/notification_service.dart:447)
#9   <asynchronous suspension>
⚠️ Could not setupFCM on Sonic Pad(ws://192.168.178.151/websocket)
💡 Getting fcm.client
[... cut ...]
⚠️ Could not retrieve key: fcm.client
💡 Show Dialog request for DialogType.logging
💡 [ws://192.168.178.151/websocket7952861] WS-Stream closed normal! Code: 1002, Reason: null
💡 [7952861-ws://192.168.178.151/websocket] ClientState.connected ➝ ClientState.disconnected
💡 [7952861]Trying to connect to ws://192.168.178.151/websocket
💡 [7952861-ws://192.168.178.151/websocket] ClientState.disconnected ➝ ClientState.connecting
💡 Using headers {}
💡 Printer Service received klippyState: KlipperState.disconnected

[image]

Bandit3112011, 2023-08-19:
@clon1998 ok?

Clon1998, 2023-08-19:
@Bandit3112011 in that case, also try the same as I described above. (Update moonraker...)

Bandit3112011, 2023-08-19:
There is no neuer Version on the Apple Store

togeis, 2023-08-19:
With latest Mobileraker and Moonraker the webcam is found 😁

closed 2023-09-05 by Clon1998