Clon1998/mobileraker #224: After canceling printing, you cannot re-select the file to print

zs96500, 2023-08-19:
After canceling the printing, you cannot select any file that needs to be printed to reprint, and the print button in the file is grayed out and cannot be used.

Clon1998, 2023-08-19:
Thats new to me, could you provide me a screen recording and the logs of the app?

The logs are located at the App's setting page at the bottom

zs96500, 2023-08-20:
[mobileraker log.zip](https://github.com/Clon1998/mobileraker/files/12387172/mobileraker.log.zip)
After canceling printing, files cannot be selected for printing.

zs96500, 2023-08-20:
> Thats new to me, could you provide me a screen recording and the logs of the app?
> 
> The logs are located at the App's setting page at the bottom

[mobileraker log.zip](https://github.com/Clon1998/mobileraker/files/12387188/mobileraker.log.zip)
After canceling printing, files cannot be selected for printing.

zs96500, 2023-08-20:
But it is possible to operate and print on the web page.

Clon1998, 2023-08-20:
Found the reason. Most print buttons only work if the printer was in the standby or complete state. This issue will be fixed in the next release!

closed 2023-09-05 by Clon1998