Clon1998/mobileraker #225: It takes 3 minutes to start the app every time

zs96500, 2023-08-19:
I am in China, every time I open this app, it takes 3 minutes to enter the app interface, but if I use a VPN to connect to the US network, the opening speed of the app is very fast, why?

Clon1998, 2023-08-19:
Because of the greate fire wall.
Some if the services I am using are blocked in your country.

However, to make the experience better for you do you mind sharing the app logs with me? They are located at the bottom of the apps settings.
Afterwards I will be able to have a look at which services are so slow and hopefully do something about it.

zs96500, 2023-08-20:
> 因为有巨大的防火墙。 如果我使用的某些服务在您所在的国家/地区被阻止。
> 
> 但是，为了让您获得更好的体验，您介意与我分享应用程序日志吗？它们位于应用程序设置的底部。 之后我将能够查看哪些服务如此缓慢，并希望对此采取措施。

[mobileraker log.zip](https://github.com/Clon1998/mobileraker/files/12387183/mobileraker.log.zip)

Clon1998, 2023-08-21:
Thanks,
sadly the logs do not contain any useful info.

Therefore, could you please send me the logs once again but after the app took 3 min to start.
Also it would be helpful if you could screen record me the startup screen since it includes Emojis that can help me to determine the Services that take so long to start.

Clon1998, 2023-08-23:
Bump.
Would be very helpful if you could send me a Screenrecording of you starting the app.

zs96500, 2023-08-23:
> Thanks, sadly the logs do not contain any useful info.
>
> Therefore, could you please send me the logs once again but after the app took 3 min to start. Also it would be helpful if you could screen [...]

zs96500, 2023-08-23:
> Bump. Would be very helpful if you could send me a Screenrecording of you starting the app.

[log and video.zip](https://github.com/Clon1998/mobileraker/files/12419151/log.and.video.zip)
Please check this file

Clon1998, 2023-08-23:
Thanks a ton for the recording!
I should have fixed the issue and will upload a new pre-release that contains a new APK that you could sideload and install. Afterward, please let me know if the loading time is reduced :).

Clon1998, 2023-08-23:
Here is the download link for the pre-release test build:

https://api.codemagic.io/artifacts/9a13a48d-db88-4129-af94-264ca50bd047/532f51ab-139c-494f-8f11-9ea9e471cae2/app-release-universal.apk

Clon1998, 2023-08-23:
Please find the pre-release here: https://github.com/Clon1998/mobileraker/releases/tag/pre-2.5.4

In case you are unable to install the app, make sure to fully remove the old app and than try to install the app again.
Please let me know if the app starts quicker now!

zs96500, 2023-08-24:
> Please find the pre-release here: https://github.com/Clon1998/mobileraker/releases/tag/pre-2.5.4
> 
> In case you are unable to install the app, make sure to fully remove the old app and then try to install the app again. Please let me know if the app starts quicker now!

I used it for about 1 hour in continuous testing, and now it starts up very quickly.

Clon1998, 2023-08-24:
> I used it for about 1 hour in continuous testing, and now it starts up very quickly.

Awesome!
I will close the issue once the new release is publicly available

zs96500, 2023-08-25:
> Awesome! I will close the issue once the new release is publicly available

In this version, when the app runs in the background for about 20 minutes, the printer cannot be connected again. You must close the app and reopen it to connect.

zs96500, 2023-08-25:
[video and log.zip](https://github.com/Clon1998/mobileraker/files/12442404/video.and.log.zip)
Here is the video and log of the problem

closed 2023-08-25 by zs96500